tinygo-org/tinygo #889: Master does not build from source on macOS

tucnak, 2020-02-04:
Hello,

I've been trying to patch `tinygo` a tiny bit, but ran into some issues off the ground.

I well-followed [Source install](https://tinygo.org/getting-started/linux/#source-install) guide for Linux. I couldn't find one for macOS, and `clang` seems to be the deal, so I figured it should be appropriate.
```
$ clang -v
Apple LLVM version 10.0.1 (clang-1001.0.46.4)
Target: x86_64-apple-darwin18.7.0
Thread model: posix

$ make llvm-source
$ make llvm-build
$ make
CGO_CPPFLAGS="-I/Users/badt/Dev/go/src/github.com/tinygo-org/tinygo/llvm-project/llvm/include -I/Users/badt/Dev/go/src/github.com/tinygo-org/tinygo/llvm-build/include -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/Users/badt/Dev/go/src/github.com/tinygo-org/tinygo/llvm-build/tools/clang/include -I/Users/badt/Dev/go/src/github.com/tinygo-org/tinygo/llvm-project/clang/include -I/Users/badt/Dev/go/src/github.com/tinygo-org/tinygo/llvm-project/lld/include" 
CGO_CXXFLAGS="-std=c++11"
CGO_LDFLAGS="/Users/badt/Dev/go/src/github.com/tinygo-org/tinygo/llvm-build/lib/libclang.a 
-std=c++11 
-L/Users/badt/Dev/go/src/github.com/tinygo-org/tinygo/llvm-build/lib  
-lclangAnalysis -lclangARCMigrate -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangStaticAnalyzerCheckers -lclangStaticAnalyzerCore -lclangStaticAnalyzerFrontend -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions  
-lstdc++  
-llldCOFF -llldCommon -llldCore -llldDriver -llldELF -llldMachO -llldMinGW -llldReaderWriter -llldWasm -llldYAML  
-L/Users/badt/Dev/go/src/github.com/tinygo-org/tinygo/llvm-build/lib
-Wl,-search_paths_first -Wl,-headerpad_max_install_names -lLLVMOption -lLLVMMCJIT -lLLVMLTO -lLLVMPasses -lLLVMObjCARCOpts -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMCoverage -lLLVMCoroutines -lLLVMipo -lLLVMInstrumentation -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMAsmParser -lLLVMAVRDisassembler -lLLVMAVRCodeGen -lLLVMAVRAsmParser -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyInfo -lLLVMRISCVDisassembler -lLLVMRISCVCodeGen -lLLVMRISCVAsmParser -lLLVMRISCVDesc -lLLVMRISCVUtils -lLLVMRISCVInfo -lLLVMAArch64Disassembler -lLLVMAArch64CodeGen -lLLVMAArch64AsmParser -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMARMDisassembler -lLLVMARMCodeGen -lLLVMARMAsmParser -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMGlobalISel -lLLVMSelectionDAG -lLLVMAsmPrinter -lLLVMDebugInfoDWARF -lLLVMCodeGen -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMX86Desc -lLLVMObject -lLLVMMCParser -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMMCDisassembler -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMDebugInfoMSF -lLLVMBinaryFormat -lLLVMX86Utils -lLLVMX86Info -lLLVMSupport -lLLVMDemangle -lm -lxml2 -lstdc++ " 
go build -o build/tinygo -tags byollvm .
# github.com/tinygo-org/tinygo/builder
Undefined symbols for architecture x86_64:
  "_Z3_ast_to_string", referenced from:
      (anonymous namespace)::Z3Expr::print(llvm::raw_ostream&) const in libLLVMSupport.a(Z3Solver.cpp.o)
  "_Z3_dec_ref", referenced from:
      (anonymous namespace)::Z3Solver::getBoolSort() in libLLVMSupport.a(Z3Solver.cpp.o)
      (anonymous namespace)::Z3Solver::getBitvectorSort(unsigned int) in libLLVMSupport.a(Z3Solver.cpp.o)
      (anonymous namespace)::Z3Solver::getFloat16Sort() in libLLVMSupport.a(Z3Solver.cpp.o)
      (anonymous namespace)::Z3Solver::getFloat32Sort() in libLLVMSupport.a(Z3Solver.cpp.o)
      (anonymous namespace)::Z3Solver::getFloat64Sort() in libLLVMSupport.a(Z3Solver.cpp.o)
      (anonymous namespace)::Z3Solver::getFloat128Sort() in libLLVMSupport.a(Z3Solver.cpp.o)
      (anonymous namespace)::Z3Solver::getSort(llvm::SMTExpr const* const&) in libLLVMSupport.a(Z3Solver.cpp.o)
      ...
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [tinygo] Error 2
```

For the record: LLVM steps passed with flying colours, but `make` just wouldn't work, and I believ it's beyond my expertise at this point to mine any more useful data out of this.

Full log can be found [here](https://gist.github.com/tucnak/13a8adaabc64fa961f4e9b75fb4487c1).

aykevl, 2020-02-05:
It definitely builds from source _on some systems_ as that is what happens in CI with each commit.
I can't really say what is going on there. What macOS version are you using? Do you have some kind of uncommon setup?

syoder, 2020-02-13:
For what it's worth, it does build for me. I appear to be on a newer version of clang:
```sh
$ clang -v
Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin19.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

deadprogram, 2020-02-15:
> I well-followed Source install guide for Linux. I couldn't find one for macOS

The instructions for a source install on macOS are located here:
https://tinygo.org/getting-started/macos/#source-install

For sure the Linux instructions do not work on macOS.

Hope that helps!

gwtnz, 2020-02-22:
I ended up using the docker image since I didn't want to pull across all the dependencies (the Mac I'm using to develop on is needed for other things, so I wanted to avoid disrupting its primary use).

To be able to adjust the source I mount the various `src`, `machine` etc folders into the docker container when building. In some cases I need to rebuild the docker container to pick up specific things, but on the whole just mounting the modified source folders works fine.

cornelk, 2020-06-30:
@tucnak it should be fixed now in the dev branch

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram