tinygo-org/tinygo #894: CompilerRT Hangs on GBA taget

trevor403, 2020-02-07:
It seems that division is still not possible on the GBA target.

Before #846 if you attempted to compile a function that used division, you would get an error due to missing arithmetic library functions from LLVM like `__umodsi3` (see https://github.com/tinygo-org/tinygo/issues/843#issuecomment-573540947)

It also seems as though GC and Optimization settings play into how well behaved code will run on GBA. I've noticed that the CompilerRT functions seem to hang the GBA program. It looks like this is due to stack corruption.

This makes it impossible to do something like `strconv.FormatUint(uint64(104), 10)`

This may also be due to the fact that 64bit data types don't fit into the registers on the GBA but I have also tried with 32bit math and get similar failures. 

I will comment back shortly with the following
- branch used
- test program
- build command

aykevl, 2020-02-07:
> This may also be due to the fact that 64bit data types don't fit into the registers on the GBA but I have also tried with 32bit math and get similar failures.

What the compiler supports and what the architecture supports is mostly separate. Example: you can use int64_t and float on AVR (with a C compiler) while the architecture supports none of that. The compiler will make sure to either split up the operation into multiple instructions (example: `add`/`adc`) or make calls to a runtime library such as compiler-rt.

> It also seems as though GC and Optimization settings play into how well behaved code will run on GBA. I've noticed that the CompilerRT functions seem to hang the GBA program. It looks like this is due to stack corruption.

I would be very surprised if it was stack corruption. Why do you think it's this?

A test program would be very much appreciated, that way I can debug this locally.

trevor403, 2020-02-07:
Sure here is an example

Here is some sample code https://github.com/trevor403/division-fail
It builds using the `tinygo/tinygo:0.12.0` docker image

trevor403, 2020-02-07:
Also note that if you switch the code at the bottom of the program to:
```
	// only use 100 to pass fastSmalls
	num := uint64(100)
	numString := strconv.FormatUint(uint64(num), 10)
	printString("num "+numString, 9)
```
This does not error unlike the loop. I think this is due to compile time optimizations for arithmetic with constants.

aykevl, 2020-02-08:
I ran the program with `tinygo gdb -target=gameboy-advance -size=short -opt=1` and found the bug:

```
(gdb) c
Continuing.

Program received signal SIGILL, Illegal instruction.
0x080009fe in strconv.formatBits (dst=..., u=<optimized out>, base=10, neg=false, append_=false) at ../../../../../../../usr/local/go/src/strconv/itoa.go:139
139				us /= 100
(gdb) list
134	
135			// u guaranteed to fit into a uint
136			us := uint(u)
137			for us >= 100 {
138				is := us % 100 * 2
139				us /= 100
140				i -= 2
141				a[i+1] = smallsString[is+1]
142				a[i+0] = smallsString[is+0]
143			}
(gdb) bt
#0  0x080009fe in strconv.formatBits (dst=..., u=<optimized out>, base=10, neg=false, append_=false)
    at ../../../../../../../usr/local/go/src/strconv/itoa.go:139
#1  0x080008d6 in strconv.FormatUint (i=429496729700, base=10) at ../../../../../../../usr/local/go/src/strconv/itoa.go:18
#2  0x08000b32 in github.com/tinygo-org/tinygo/tmp/division-fail.main () at division-fail/main.go:31
```

Disassembling, I get:

```
   │0x80009f6 <strconv.formatBits+282>      movs   r1, r4
   │0x80009f8 <strconv.formatBits+284>      movs   r7, r3
   │0x80009fa <strconv.formatBits+286>      blx    0x80011b8 <__udivsi3>
  >│0x80009fe <strconv.formatBits+290>      movs   r3, r7
   │0x8000a00 <strconv.formatBits+292>      movs   r1, r4
   │0x8000a02 <strconv.formatBits+294>      muls   r1, r0
   │0x8000a04 <strconv.formatBits+296>      subs   r1, r5, r1
```

I think this is inserted by the linker. Remember that warning you get each time you build?

```
ld.lld-9: warning: lld uses blx instruction, no object with architecture supporting feature detected
```

trevor403, 2020-02-08:
Yes I am aware of that warning -- good catch :+1:

deadprogram, 2020-02-09:
Looks like the current operative solution is to use a different linker until if/when lld addresses the core issue.

This is now the Rust Embedded team looks to have handled:
https://github.com/rust-lang/stdarch/commit/2fdd3ac5852ca5474afae0a11214c54a0c4bd11a

aykevl, 2020-02-09:
I tried, but it doesn't seem to work. This is what I did:

```diff
diff --git a/targets/gameboy-advance.json b/targets/gameboy-advance.json
index 8e7fee8b..9b829f31 100644
--- a/targets/gameboy-advance.json
+++ b/targets/gameboy-advance.json
@@ -5,7 +5,7 @@
        "goos": "linux",
        "goarch": "arm",
        "compiler": "clang",
-       "linker": "ld.lld",
+       "linker": "arm-none-eabi-ld",
        "rtlib": "compiler-rt",
        "cflags": [
                "-g",
```

Maybe it's not the linker.

trevor403, 2020-02-10:
It removes all unsupported `blx` calls with labels instead of registers. It seems to work for me with the above patch. 

However this program still does not run successfully.

trevor403, 2020-02-10:
I was able to avoid encountering a division error by writing a Fixed base `FormatUint` https://github.com/trevor403/division-fail/commit/2c05086e4991bd8d14ddadbeaa8b0c7444d32960

However this issue still needs to be fixed in order to support division on GBA. I can make a PR to add devkitARM to the docker image if you think that would be a good idea @deadprogram

aykevl, 2020-02-11:
> I can make a PR to add devkitARM to the docker image if you think that would be a good idea @deadprogram

Does that fix the issue?
I would be surprised if it does, considering that `arm-none-eabi-ld` is a much older and more mature linker and is unlikely to have such bugs. I think it's more likely that there is a bug in TinyGo, LLVM, or mGBA.

trevor403, 2020-02-11:
Yes as I stated above it causes all instances of `blx {label}` when using `lld.ld` to change to `bl {label}`. I confirmed with objdump. 

However there is still a crash I am looking into.

jaddr2line, 2020-04-19:
While testing #1060, I found that a `blx` was being used to branch into the `gba-display` example's `main` function when using the coroutines scheduler. It uses a supported branch instruction instead when using `-scheduler=none` or when an infinite loop is added to the end of the function (making it `noreturn`).

aykevl, 2020-04-26:
This issue has been fixed in #1067.
[image]

closed 2020-04-26 by aykevl

mogenson, 2020-11-29:
Hey, I'm still seeing this `blx` symbol being inserted. The resulting GBA ROM crashes in MGBA and a physical GBA.

I'm on commit 0c4f0b1ebf123a770f273bbed29487685430d404 of the dev branch, using Arch Linux and LLVM 11. Here's what I'm trying:

```
$ go build -tags=llvm11 -o build/tinygo
$ make gen-device
$ ./build/tinygo build -size=full -target=gameboy-advance -o build/gba-display.gba src/examples/gba-display/gba-display.go 
ld.lld: warning: lld uses blx instruction, no object with architecture supporting feature detected
   code  rodata    data     bss |   flash     ram | package
    244       0       0       0 |     244       0 | (bootstrap)
     88       0       0       0 |      88       0 | command-line-arguments
      0       0       0      16 |       0      16 | runtime
    332       0       0      16 |     332      16 | (sum)
    604       -       0    3092 |     604    3092 | (all)
$ mgba build/gba-display.gba
The game crashed!
```

I also gave the docker image a try:
```
$ docker pull tinygo/tinygo:latest
$ docker run --rm -v $(pwd):/src tinygo/tinygo:latest tinygo build -o /src/build/gba-display.gba -size=full -target=gameboy-advance examples/gba-display
tinygo:ld.lld: warning: lld uses blx instruction, no object with architecture supporting feature detected
   code  rodata    data     bss |   flash     ram | package
    244       0       0       0 |     244       0 | (bootstrap)
     88       0       0       0 |      88       0 | examples/gba-display
      0       0       0      16 |       0      16 | runtime
    332       0       0      16 |     332      16 | (sum)
    604       -       0    3092 |     604    3092 | (all)
$ mgba build/gba-display.gba
The game crashed!
```

Let me know if there's anything I can try. Thanks!

aykevl, 2020-12-01:
What you are seeing here is a bug in the generation of `.gba` files. No `blx` instruction is created.

You can verify this yourself. First see that the program actually works:

    tinygo run -target=gameboy-advance examples/gba-display

Then by inspecting the assembly:

    tinygo build -o test.elf -target=gameboy-advance examples/gba-display
    arm-none-eabi-objdump -d test.elf | grep blx

This line:

    tinygo:ld.lld: warning: lld uses blx instruction, no object with architecture supporting feature detected

is just a warning. It doesn't actually create such an instruction, it only warns that it _could_ create such an instruction: lld doesn't fully support this old ARM architecture. But I think it will only create this instruction when calling between ARM and Thumb code, not with regular (ARM only) calls. This is why #1067 fixes this issue: it avoids all Thumb code to work around this linker bug (or should I say, missing feature?).

mogenson, 2020-12-03:
Hey, you're totally right. My mistake. Thanks for the info.

If I specify `elf` as the output extension, mgba runs the compiled `elf` perfectly. It seems that something may be off with the binary ROM header and that is why mgba and a physical GBA are crashing. To test, I copied the header from this Rust project's [startup assembly](https://github.com/rust-console/gba/blob/master/crt0.s).

Before:
```
$ ./build/tinygo build -target=gameboy-advance -o build/gba-display.gba examples/gba-display
$ file build/gba-display.gba
build/gba-display.gba: data
$ mgba build/gba-display.gba
The game crashed!
```

After:
```
$ ./build/tinygo build -target=gameboy-advance -o build/gba-display.gba examples/gba-display
$ file build/gba-display.gba
build/gba-display.gba: Game Boy Advance ROM image (, Rev.00)
$ mgba build/gba-display.gba
# running successfully...
```

Should I create another issue about the GBA ROM header or would you be open to a PR?

aykevl, 2020-12-07:
Yes, a PR would definitely be welcome!

trevor403, 2021-01-04:
> It seems that something may be off with the binary ROM header and that is why mgba and a physical GBA are crashing.

Yes you are meant to run `gbafix` on the GBA file to recover the Nintendo Logo which is a form of copy protection on GBA Roms.

It should be safe to include logo bytes in the startup code at `targets/gameboy-advance.s` instead of filling it with zeros.

PR #1067 (2020-04-21): gba: always use ARM mode instead of Thumb mode
This results in bigger code size, but it works around a bug in the linker.

The issue starts with the problem that libraries (picolibc, compiler-rt) were compiled as ARM and the rest as Thumb. This causes some blx instructions to be inserted by the linker to call into these libraries.

Ideally we should fix the libraries to use Thumb mode instead, but that requires some more extensive changes (including fixes to compiler-rt) and it's just way easier to use ARM mode everywhere.

fixes #894