tinygo-org/tinygo #903: How to use Flash ROM in Arduino

inajob, 2020-02-16:
How to use Flash ROM in Arduino?
I'm trying to migrate font data from https://github.com/adafruit/Adafruit-GFX-Library/blob/master/glcdfont.c .

This code use PROGMEM.
But tinygo doesn't support PROGMEM.
I try to use "const" but const array intialization seems not to be supported in Go.
(https://stackoverflow.com/questions/13137463/declare-a-constant-array )

How do I use this font data ? ( I think this is common in Arduino Uno Project)

aykevl, 2020-02-18:
There is no way to do this at the moment.

Unlike for example ARM that has a single address space, AVR has two different address spaces. I think there are ways to determine (at compile time) in which address space a global should be, but it will only ever work in a limited number of cases. Even C, which technically does support multiple address spaces, has trouble with PROGMEM and requires lots of custom annotations to work correctly.

One way to solve this would be the following proposal: https://github.com/golang/go/issues/35950

inajob, 2020-02-18:
I understood.
I'm excited to write program for AVR with Go.
I look forwad to the day when tinygo(or go) can deal with different address spaces.

closed 2020-02-18 by inajob

conejoninja, 2020-02-21:
tinyfont ( https://github.com/tinygo-org/tinyfont ) and tinydraw ( https://github.com/tinygo-org/tinydraw ) are based in adafruit gfx library, while that font is not ported yet, "proggy" might be a good alternative if that's what you are looking for.