tinygo-org/tinygo #909: Windows 10, can’t compile the project.

evseevt, 2020-02-21:
OS : Windows 10

aykevl, 2020-02-22:
What are you trying to do?

TinyGo does not yet support Windows as a compiler target. The compiler runs on Windows, but it cannot yet create Windows binaries. And this is what TinyGo defaults to with no other flags.

If you want to compile a binary for a microcontroller board or for webassembly, use the `-target` flag.

evseevt, 2020-02-23:
I'm using Windows and WSL (Ubuntu), command "tinygo build -o main/main.hex -target arduino main/main.go", and get .hex file compile correctly =). 

But in Proteus simulator [http://bit.do/fwhNt](url), generated .hex file (tinygo) don't work.. Arduino-IDE compile .hex work's fine. It problem in Proteus or in tinygo?

I want upload tinygo .hex file into Arduino Proteus simulator...
**Video demonstration** : [https://youtu.be/omDvPYhQCMU](url)

evseevt, 2020-02-23:
> What are you trying to do?
>
> TinyGo does not yet support Windows as a compiler target. The compiler runs on Windows, but it cannot yet create Windows binaries. And this is what TinyGo defaults to [...]
I'm using Windows and WSL (Ubuntu), command "tinygo build -o main/main.hex -target arduino main/main.go", and get .hex file compile correctly =).

But in Proteus simulator http://bit.do/fwhNt, generated .hex file (tinygo) don't work.. Arduino-IDE compile .hex work's fine. It problem in Proteus or in tinygo?

I want upload tinygo .hex file into Arduino Proteus simulator...
Video demonstration : https://youtu.be/omDvPYhQCMU

evseevt, 2020-02-23:
TinyGo code:

package main

// This is the most minimal blinky example and should run almost everywhere.

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
}

closed 2020-02-23 by evseevt

aykevl, 2020-02-24:
I think that Protheus is not able to correctly simulate the binaries produced by TinyGo. They work fine for me on real hardware.
I have attached a hex file that definitely works (renamed from test.hex to test.hex.txt for GitHub).

[test.hex.txt](https://github.com/tinygo-org/tinygo/files/4244722/test.hex.txt)

evseevt, 2020-02-24:
> I think that Protheus is not able to correctly simulate the binaries produced by TinyGo. They work fine for me on real hardware.
> I have attached a hex file that definitely works (renamed from [...]

This problem in Proteus, I uploaded in my real board .hex from Ubuntu and Windows compiler, TinyGo .hex file works corretly in board. This question is closed, thanks for the support.