tinygo-org/tinygo #912: Calling yield from sleepTicks causes a stack overflow, but panics on nil pointer dereference

firelizzard18, 2020-02-24:
I am working on implementing a target for the Teensy 3.6 / NXP MK66 (#863). My initial implementation of `sleepTicks` called `Gosched`. When there is only one user routine, and that routine calls `time.Sleep` (`runtime.sleep`), this will cause a stack overflow (`sleep` => `yield` => `sleepTicks` => `Gosched` => `yield` => ...). In this particular scenario, `currentTask` becomes `nil` and `yield` panics.

Obviously `sleepTicks` should not call `Gosched` or `yield`. But if that happens, `fatal error: stack overflow` is the proper response, not `panic: runtime error: nil pointer dereference`.

https://github.com/tinygo-org/tinygo/blob/c0b8716d7f2cf33bcbb7e452e84260228a4e2b32/src/runtime/scheduler_tasks.go#L100-L107

```golang
// sleepTicks spins for a number of microseconds
func sleepTicks(d timeUnit) {
	// TODO actually sleep

	if d <= 0 {
		return
	}

	start := ticks()
	ms := d / 1000

	for {
		for ticks()-start >= 1000 {
			ms--
			if ms <= 0 {
				return
			}
			start += 1000
		}
		Gosched()
	}
}
```

jaddr2line, 2020-02-24:
You cannot call yield from inside sleepticks. It must be called from inside a goroutine.

firelizzard18, 2020-02-24:
@jaddr2line Yes, I understand that now, as I said:

> Obviously sleepTicks should not call Gosched or yield.

As I also said:

> But if that happens, `fatal error: stack overflow` is the proper response, not `panic: runtime error: nil pointer dereference`.

This issue is not about my incorrect code. This issue is about a secondary error that hid the real error in my code.

jaddr2line, 2020-02-24:
There is no stack overflow involved. There is no recursion. The `yield` call is supposed to switch back to the scheduler, which then continues doing whatever. This behavior is correct. You cannot call `yield` except from a goroutine, and in this case it is called from the scheduler.

jaddr2line, 2020-02-24:
See: https://github.com/tinygo-org/tinygo/blob/c0b8716d7f2cf33bcbb7e452e84260228a4e2b32/src/runtime/scheduler.go#L266

closed 2020-02-24 by jaddr2line

firelizzard18, 2020-02-24:
If there is no stack overflow, why does that error appear?

jaddr2line, 2020-02-24:
The nil pointer error appears because there is no current goroutine when on the scheduler

firelizzard18, 2020-02-24:
So then the issue is "Calling yield in the scheduler causes a nil pointer dereference," with the solution being `if currentTask == nil { panic("called yield from within scheduler") }`

aykevl, 2020-02-24:
I agree with @jaddr2line here. `sleepTicks` is specific to the runtime and thus has more restrictions than most functions. It is not an exported API.

I agree this is confusing but I think the best way to solve this is better documentation how to port TinyGo to a new chip family (explaining initialization, `putchar`, `tickMicros`, `sleepTicks`, etc).

firelizzard18, 2020-02-24:
I agree that TinyGo needs better documentation on adding a new chip family, but the first line of defence for documentation should always be meaningful errors. Thus `yield` should panic with (some variation of) `called yield from within scheduler` and not `runtime error: nil pointer dereference`.

Adding `if currentTask == nil { ... }` does not add anything to the fast path. Without the extra check, the compiler must insert its own check before `*currentTask` to guard against a nil pointer. In either case, the fast path is:

```asm
push	{r4, lr}
movw	r4, <addr>
movt	r4, <addr>
ldr	r0, [r4, #0]
; branch - check for nil
ldr	r1, [r0, #52]
; branch - check for nil
ldr	r1, [r1, #0]
movw	r2, <canary>
movt	r2, <canary>
cmp	r1, r2
; branch - check for nil
ldmia.w	sp!, {r4, lr}
b.w	tinygo_switchToScheduler
```

aykevl, 2020-02-25:
> Adding `if currentTask == nil { ... }` does not add anything to the fast path.

But it does increase code size, which is a problem.

What you can do, is do the check only while debugging the scheduler:

```go
if schedulerDebug && currentTask == nil {
    runtimePanic("called yield while currentTask is nil")
}
```

This check should be optimized out in regular builds.

jaddr2line, 2020-02-25:
Also worth testing with the `coroutines` scheduler (`-scheduler coroutines`) for compatibility. This would not compile with `coroutines`.

firelizzard18, 2020-02-25:
Adding `schedulerDebug` to the check makes sense to me, since someone using an existing target should never see this situation. Because I wanted to know, adding `if currentTask == nil { panic(...) }` increases the binary size by 44 bytes plus the size of the error message (I tested with an empty string), and with `schedulerDebug && ...`, that path is indeed optimized out in regular builds.

> This would not compile with `coroutines`.

I'm not sure what you mean. Blinky compiles and works with or without `-scheduler coroutines` and with or without the additional check. In fact, if I switch back to using `Gosched()` in `sleepTicks`, it works fine using the coroutines scheduler. I don't plan on actually doing that, beyond satisfying my curiosity, but it is interesting that it doesn't fail with the other scheduler.