tinygo-org/tinygo #918: -target arduino "PinInputPullup not declared by package machine".

evseevt, 2020-02-25:
Traget board : Arduino UNO.
Goal : Up-Down -> "Pull-up Resistor".
Result build : PinInputPullup not declared by package machine.

Code :
```
package main

import (
	"machine"
	"time"
)

const (
	led    = machine.Pin(13)
	button = machine.Pin(12)
)

func main() {
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	button.Configure(machine.PinConfig{Mode: machine.PinInputPullup})

	for {
		if !button.Get() {
			led.High()
		} else {
			led.Low()
		}

		time.Sleep(time.Millisecond * 10)
	}
}
```
Comment : Arduino UNO don't support Pull-up Resistors?

deadprogram, 2020-02-26:
Many AVR processors support a built-in pullup resistor. This would just require some additional code, which we would be most pleased to accept should someone happen to submit a PR with this feature.

deadprogram, 2020-05-22:
Support for this feature was just merged into the `dev` branch as #1109 and will be part of the next release.

bobross419, 2020-07-27:
This (PinInputPullup) worked for me on arduino-mega2560 R3.  I was finally able to read digital button inputs.  Thanks for putting this together!

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram

yevsia, 2020-08-15:
Thanks, works fine)