tinygo-org/tinygo #920: Consider adding functions for sleep, stop, etc

firelizzard18, 2020-02-26:
Cortex-M chips support multiple power levels. Should TinyGo add functions to transition into these modes? The SDK that Freescale/NXP provides for the K66 (the chip the Teensy uses) includes `wait` (aka sleep) and `stop`, which I ported to (Tiny)Go:

```golang
// Stop enters STOP (deep sleep) mode
func Stop() {
	// set SLEEPDEEP to enable deep sleep
	nxp.SystemControl.SCR.SetBits(nxp.SystemControl_SCR_SLEEPDEEP)

	// enter STOP mode
	arm.Asm("wfi")
}

// Wait enters WAIT (sleep) mode
func Wait() {
	// clear SLEEPDEEP bit to disable deep sleep
	nxp.SystemControl.SCR.ClearBits(nxp.SystemControl_SCR_SLEEPDEEP)

	// enter WAIT mode
	arm.Asm("wfi")
}
```

Is it valuable to provide these functions as part of `machine` or `runtime` (or another package)?

deadprogram, 2020-02-26:
They for sure seem useful.

The functions that apply to all of the various ARM processors have been getting put here https://github.com/tinygo-org/tinygo/blob/master/src/device/arm/arm.go

Not sure if these functions are specific to NXP, in which case they probably would need to get put into some new file in `src/device/nxp" that is intended to be persisted.

What do you think?

aykevl, 2020-02-26:
I do not think the sleep/stop should be directly exposed: the scheduler should automatically go into a low power state once all goroutines are blocked (which is currently only happening with `time.Sleep`). For example, see this piece of code:

https://github.com/tinygo-org/tinygo/blob/1d913a62bc0ceef841953058c2c6fb1dd2a40f6a/src/runtime/runtime_nrf.go#L101-L103

What does the `SLEEPDEEP` bit change? Wakeup latency?

firelizzard18, 2020-02-26:
I’m on the road so I can’t look up the details, but I’m pretty sure STOP and WAIT mode are universal Cortex-M features. The K66 has around 10 different power modes, but I believe STOP and WAIT are generic.

I agree that sleeping/WAIT mode probably should remain a scheduler detail. However, STOP/deep sleep is a lower power state that further reduces current draw and usually disables some peripherals. That definitely shouldn’t be done automatically. A device that spends most of its time quiescent and needs to occasionally wake up on interrupt to do something would benefit from deep sleeping while it was inactive, especially if it’s battery powered.

I’ll look up the relevant specs and post that here when I can.

firelizzard18, 2020-02-27:
ARM's documentation shows a SLEEPDEEP bit for Cortex-M0, M3, and M4 (I didn't look at the others). The M0 spec has the comment, "If your device does not support two sleep modes, the effect of changing the value of this bit is implementation-defined." The M3 and M4 spec do not have this comment. In each case the field description is "Controls whether the processor uses sleep or deep sleep as its low power mode."

The K66 in particular has 11 low power modes, described in table 8-1 in section 8.3 "Power Modes Description". Normal wait (WFI with SLEEPDEEP=0) "Allows peripherals to function while the core is in sleep mode, reducing power. NVIC remains sensitive to interrupts; peripherals continue to be clocked." Normal stop (WFI with SLEEPDEEP=1) "Places chip in static state. Lowest power mode that retains all registers while maintaining LVD protection. NVIC is disabled; AWIC is used to wake up from interrupt; peripheral clocks are stopped." The other 9 low power modes have various caveats and are not generic as far as I know, so I would expect developers utilizing those to be using the spec to write chip-specific code.

aykevl, 2020-02-27:
All I read about this is that it is highly chip-specific. Some just slow down a clock which in turn makes interrupts slower (but the system works identical otherwise), others actually shut down part of the chip. This distinction is important.

I see roughly three classes of low power modes:

  * Part of the system gets shut down with the result that interrupt latency is higher, but no other visible effects. This is an application tradeoff and should be easy to implement: just provide a function to set the desired sleep depth.
  * Part of the system shuts down which results in some peripherals losing state or not functioning (for example, a UART may not trigger an interrupt anymore on incoming data). This is highly system specific. It seems like the sleep modes of the NXP chip fall in this category. I'm not quite sure how this should best be implemented, I would like to see actual use cases before deciding on an API.
  * The whole system shuts down and will wake up with a reset (that means RAM is also cleared). Examples are System-OFF in the nrf series and deep sleep in the ESP series. Perhaps this could be modeled with a `os.Exit` call, as it effectively shuts down the system.