tinygo-org/tinygo #921: llvm-ar & llvm-nm "not found"

justinclift, 2020-02-26:
Looking at TinyGo again, compiling from the `dev` branch now shows:

```
$ make
Makefile:25: llvm-ar not found
Makefile:28: llvm-nm not found
...
```

This is when compiling on Linux, using LLVM compiled from source in the `llvm-project` and `llvm-build` directories.  `llvm-ar` and `llvm-nm` are definitely present as expected:

```
$ ls -l llvm-build/bin/llvm-ar* llvm-build/bin/llvm-nm*
-rwxr-xr-x 1 jc users 11327528 Feb 18 07:14 llvm-build/bin/llvm-ar
-rwxr-xr-x 1 jc users 11460088 Feb 18 07:35 llvm-build/bin/llvm-nm
```

Looks like this warning started with 94ec082f6ca6a661ccaa88c03f80b5eb1aa53d88.  Not sure (yet) if it's indicating a real problem or not, as the `tinygo` binary is being generated ok.

justinclift, 2020-02-26:
@jaddr2line This might be of interest. :smile:

aykevl, 2020-02-26:
Maybe it's best to not show a warning when llvm-ar or llvm-nm are not found. It will lead to a pretty clear error message anyway as soon as those commands are needed. Something like:

```make
ifneq (, $(shell command -v llvm-nm-9 2> /dev/null))
    LLVM_NM ?= llvm-nm-9
else
    LLVM_NM ?= llvm-nm
endif
```

With the added support for `llvm-build/bin/llvm-ar` etc:

```make
ifneq (, $(shell command -v llvm-nm-9 2> /dev/null))
    LLVM_NM ?= llvm-nm-9
else ifneq (, $(shell command -v llvm-build/bin/llvm-nm 2> /dev/null))
    LLVM_NM ?= llvm-build/bin/llvm-nm
else
    LLVM_NM ?= llvm-nm
endif
```

justinclift, 2020-02-27:
Yeah, I'm not (yet) sure of the better approach either, due to #925.

On the other hand, if we do add the `llvm-build/bin/` directory version, that would probably make for a more robust solution for solving #925.

aykevl, 2020-02-27:
Should be fixed with https://github.com/tinygo-org/tinygo/pull/928.

aykevl, 2020-02-28:
This should now be fixed in the dev branch, so closing.

closed 2020-02-28 by aykevl