tinygo-org/tinygo #922: Problem with cmsis-svd submodule not updating

justinclift, 2020-02-26:
There seems to be a problem with the `cmsis-svd` submodule definition:

```
$ git submodule update
error: Server does not allow request for unadvertised object 15b462f152af86f3d15b952e1a5cf1bb9e2693e8
Fetched in submodule path 'lib/cmsis-svd', but it did not contain 15b462f152af86f3d15b952e1a5cf1bb9e2693e8. Direct fetching of that commit failed.
```

This is 100% reproducible on a new git checkout of the repo:

```
$ git clone --recursive https://github.com/tinygo-org/tinygo    (this grabs the master branch)
...
$ cd tinygo
$ git checkout dev
M       lib/cmsis-svd
Branch 'dev' set up to track remote branch 'dev' from 'origin'.
Switched to a new branch 'dev'
$ git submodule update
error: Server does not allow request for unadvertised object 15b462f152af86f3d15b952e1a5cf1bb9e2693e8
Fetched in submodule path 'lib/cmsis-svd', but it did not contain 15b462f152af86f3d15b952e1a5cf1bb9e2693e8. Direct fetching of that commit failed.
```

aykevl, 2020-02-26:
I think I've seen such an issue before. The following works for me (switching the branch before pulling submodules):

```
$ git clone https://github.com/tinygo-org/tinygo
$ cd tinygo
$ git checkout dev
$ git submodule update --init
```

I suspect git doesn't automatically pick up the new upstream repository. Therefore, this should be automatically fixed with the next release.

justinclift, 2020-02-27:
Thanks @aykevl, that works. :smile:

closed 2020-02-27 by justinclift