tinygo-org/tinygo #923: Wasm examples no longer compile

justinclift, 2020-02-26:
Discovered a problem (and solution) earlier on today, when attempting to compile the TinyGo wasm examples in under the `src/examples/wasm` directory:

```
$ cd src/examples/wasm
$ make main
rm -rf ./html
mkdir ./html
cp ../../../targets/wasm_exec.js ./html/
tinygo build -o ./html/wasm.wasm -target wasm -no-debug ./main/main.go
tinygo:wasm-ld: error: cannot open /home/jc/go/src/github.com/tinygo-org/tinygo/lib/wasi-libc/sysroot/lib/wasm32-wasi/libc.a: No such file or directory
error: failed to link /tmp/tinygo685543697/main: failed to link using built-in wasm-ld
make: *** [Makefile:23: main] Error 1
```

The problem turned out to be that the wasi-libc submodule needs to be (manually) compiled first, before the compiling the wasm examples:

```
$ cd lib/wasi-libc
$ make WASM_CC=~/go/src/github.com/tinygo-org/tinygo/llvm-build/bin/clang \
  WASM_AR=~/go/src/github.com/tinygo-org/tinygo/llvm-build/bin/llvm-ar \
  WASM_NM=~/go/src/github.com/tinygo-org/tinygo/llvm-build/bin/llvm-nm
```

This compile step creates the `libc.a` file being complained about in the initial error.

The proper solution to this might be something like including this in the top level Makefile, so it's done automatically during the normal TinyGo compilation.

aykevl, 2020-02-27:
This is only a problem with local builds of TinyGo. Release builds have everything set up correctly.

You can run `make wasi-libc` in the TinyGo root directory, after possibly setting `WASM_CC` etc. https://github.com/tinygo-org/tinygo/pull/928 should make this easier.

aykevl, 2020-02-28:
Maybe this should be included in the documentation? For example, just after https://tinygo.org/getting-started/linux/#additional-requirements-for-microcontrollers.

Another option could be to build it when needed, similar to how compiler-rt is currently built when needed. That depends on https://github.com/tinygo-org/tinygo/pull/871 however.

justinclift, 2020-02-28:
Adding it to the docs is probably a good starting point. :smile:

deadprogram, 2020-03-16:
Agreed that this needs to be in the docs.

deadprogram, 2020-05-13:
I think this has now been addressed, so closing. Please reopen if needed, thanks!

closed 2020-05-13 by deadprogram

tve, 2020-07-28:
How has this been addressed? I'm running arch linux (manjaro) and followed the getting started > linux > arch directions, which boil down to installing the tinygo-bin AUR package. The result is:
```
> tinygo build -o hello-world.wasm -target=wasm hello-world.go
tinygo:wasm-ld: error: /usr/lib/tinygo/lib/wasi-libc/sysroot/lib/wasm32-wasi/libc.a: archive has no
index; run ranlib to add one
failed to run tool: wasm-ld
```
Not sure what to do next...

justinclift, 2020-07-28:
@tve In theory, running `make wasi-libc` should set up the WASI piece for you.  At least, it did when this issue was filed. :smile:

tve, 2020-07-28:
After something like `pamac tinygo-bin` where would I type `make wasi-libc`?
FYI, there's a `tinygo` package https://www.archlinux.org/packages/?sort=&q=tinygo&maintainer=&flagged= that works correctly.

justinclift, 2020-07-28:
Good question. :grin:

My first thought was the tinygo package might need updating to include the WASI step in it's build process.

However, looking the file list for it already includes compiled wasi files:

https://www.archlinux.org/packages/community/x86_64/tinygo/files/

(eg search for `wasi-libc/sysroot/lib/wasm32-wasi` to see them)

Now I'm kind of wondering if there's a different step the package still needs to do, so the wasi pieces were found.

Note that I'm not that familiar with Arch personally. :shrug:

justinclift, 2020-07-28:
Reading over your error message again:

```
tinygo:wasm-ld: error: /usr/lib/tinygo/lib/wasi-libc/sysroot/lib/wasm32-wasi/libc.a: archive has no
index; run ranlib to add one
failed to run tool: wasm-ld
```

Maybe the `/usr/lib/tinygo/lib/wasi-libc/sysroot/lib/wasm32-wasi/libc.a` file included in the package wasn't created 100% correctly, and running the `ranlib` command (as per error message) on it might help?

justinclift, 2020-07-28:
Doing a general search online for that error message is throwing up lots of possibilities.

I'd tend to explore those next, unless someone with good idea(s) chimes in here. :smile:

justinclift, 2020-07-28:
This looks relevant: https://github.com/emscripten-core/emscripten/issues/9329#issuecomment-525445834

aykevl, 2020-07-29:
This looks like a problem with the package. This should work out of the box, no modification needed.

@johanbrandhorst do you have an idea what's going on?

johanbrandhorst, 2020-07-30:
The `tinygo-bin` package just downloads the latest release binary, it doesn't attempt to build anything. It's possible it's missing some runtime dependencies? It currently declares `gcc-libs` and `go` dependencies, should we be requiring anything else to run the rebuilt binaries?

aykevl, 2020-07-30:
Strange that it fails just on Arch. @tve to be absolutely sure the packaging didn't mess something up, can you try whether the prebuilt binaries from the release page don't have this error?
https://github.com/tinygo-org/tinygo/releases/tag/v0.13.1

tcastelly, 2021-05-03:
Thank you @aykevl
I'm Archlinux user with the same error. Build my app from the downloaded release works for me.

> Strange that it fails just on Arch. @tve to be absolutely sure the packaging didn't mess something up, can you try whether the prebuilt binaries from the release page don't have this error?
> https://github.com/tinygo-org/tinygo/releases/tag/v0.13.1

aykevl, 2021-05-05:
Okay, then I would consider this a bug in the `tinygo` Arch Linux package, not in TinyGo itself.

johanbrandhorst, 2021-05-05:
@tcastelly could you figure out the error and report it on the relevant package? I'd be happy to take contributions.