tinygo-org/tinygo #927: Usage within GoLand IDE no longer functional?

imle, 2020-02-27:
I found a PR that suggested I would be able to point GoLand to a GOROOT of tinygo, but I was unable to get that working. Does anyone currently use GoLand to develop with tinygo? I'd be willing to put together documentation for setting it up if it does work.

aykevl, 2020-02-27:
I don't use GoLand, I use VS Code with the Go extension or vim with a Go extension.

> I found a PR that suggested I would be able to point GoLand to a GOROOT of tinygo, but I was unable to get that working.

What exactly was the result?
You will probably also need to add a load of build tags, you can get a list of build tags with (for example) `tinygo info microbit`.

bgould, 2020-02-27:
I use Intellij, which is essentially the same as GoLand.  Here is how I set it up:

 * Disable Go Modules/vgo integration
 * Set GOROOT to point at your Go installation
 * Set up 2 GOPATH entries - the first should point at your actual GOPATH and the second should point at your tinygo installation (this second entry allows the "machine" package to be resolved properly)
 * Under Build Tags & Vendoring, set Custom Tags by pasting the values that are output from `tinygo info -target <your target board>` as @aykevl mentioned above

For me, that is enough to make it so that error highlighting and code completion works properly, even with the TinyGo-specific packages such as "machine" or "device/arm".  I keep meaning to add this to the wiki but I never get around to it.

Hope that helps.

justinclift, 2020-03-01:
>  I keep meaning to add this to the wiki but I never get around to it.

Any chance of getting that done? :smile:

artspb, 2020-03-12:
I wrote a post on the topic, you can find it [here](https://artspb.me/posts/working-with-tinygo-in-goland/). If something doesn't work for you or you'd like to suggest an improvement for GoLand, please don't hesitate to share your feedback.

bus710, 2020-04-13:
Hi, 

I have some issue with Vscode to use it with Tiny-go. 

VSCODE has some feature to allow to have additional GOPATH by using **./vscode/settings.json** per workplace so that I tried couple of options:
```json
{
    // general configuration for go extension
    "workbench.startupEditor": "newUntitledFile",
    "workbench.sideBar.location": "right",
    "editor.minimap.enabled": false,
    "workbench.colorTheme": "Dracula",
    "window.menuBarVisibility": "toggle",
    "go.formatTool": "goimports",
    "go.useLanguageServer": true,
    "git.enableSmartCommit": true,
    "git.autofetch": true,
    "[go]": {},

    // specific configuration for GOROOT and GOPATH
    "go.goroot": "/usr/local/go",
    "go.gopath": "${env:HOME}/go:/usr/local/tinygo",
}
```

Having the 2 lines at the end of the configuration didn't help me yet (Couldn't import the package! warning)

I also tried couple more lines such as:
```json
...
    "go.toolsGopath": "${env:HOME}/go:/usr/local/tinygo",
    "go.toolsEnvVars": {
        "GOROOT": "/usr/local/go",
        "GOPATH":"${env:HOME}/go:/usr/local/tinygo",
    }
}
```

...didn't work as well.

From my understanding, I shouldn't change GOROOT to use the tools like gopls, goimports, and so on but GOPATH should have the two locations - $HOME/go and /usr/local/tinygo. However, I still get the warning - Couldn't import machine and time packages when the blinky1 example is being edited. Adding the **replace** keyword in **go.mod** to point to the package's local location only removes the warning but still the code completion feature doesn't work.

I currently use these combination:
- Ubuntu 18.4 (technically mint...)
- Go 1.13
- Tiny-go 0.12
- Vscode 1.43.2
- Vscode Go extension 0.13.1

Tiny-go itself works very well to compile for MCUs in terminal so that I think I can safely say that this is an issue of my setting in Vscode. 

Any idea?

hyangah, 2020-04-13:
@bus710 I didn't try tinygo myself, but based on the awesome blog post by @artspb and the comment from @bgould, it seems build tags and GOOS/GOARCH settings are needed too. 
`go.buildTags` to add buile tags, `go.toolsEnvVars` to teach underlying tools about GOOS/GOARCH env vars. 

And, I am not sure if tinygo supports modules yet. (correct me if it does - I was just reading the above comments).

I am not sure if that's the end of the story though.  Please try and post how it goes.

bus710, 2020-04-13:
@hyangah, Thanks for your advice.
I tried GoLand to have some idea as guidance of @artspb and @bgould. 
GoLand works great as long as I don't forget these: 
1. uncheck the option "index entire GOPATH" and pick "NO SDK" when I create a project.
2. put right places for tinygo and the project itself (but index entire GOPATH should be checked this time)
3. put right tags in the **build tags > custom tags** input field as the info sub command spits out
Then, GoLand completes when I put a dot.

Btw, the build tags for nano33 iot are:
```
$ tinygo info -target arduino-nano33-iot
LLVM triple:       arduino-nano33-iot
GOOS:              iot
GOARCH:            arduino
build tags:        iot arduino tinygo gc.conservative scheduler.coroutines
garbage collector: conservative
scheduler:         coroutines
```

Now, I also tried VSCODE with these in the settings.json of workspace:
```json
...
    "[go]": {},
    "go.goroot": "/usr/local/go",
    "go.gopath": "/usr/local/tinygo:${env:HOME}/go",
    "go.buildTags": "iot:arduino:tinygo:gc.conservative:scheduler.coroutines",
    "go.toolsEnvVars": {
        "GOARCH": "arm",
        "GOOS": "linux",
    }
}
```
Some thoughts:
- for buildTags, not sure if that is right format to put several tags in a line.
- for GOARCH, I used arm because arduino for that makes gopls crashes
- for GOOS, I also used linux because iot for that makes gopls crashes (and GoLand doens't come with iot for GOOS)

With the above config, it doesn't show red under lines for the packages but the LED of machine shows the warning - "LED not declare by package machine" which seems like the tag "arduino" doesn't work in this case.

Any help appreciated,

hyangah, 2020-04-13:
@bus710 Thanks for the updates and good to hear that goland works well!

I just realized you are using `gopls` (the go language server VS Code Go uses) and there is a pending issue about propagating the build tags. https://github.com/microsoft/vscode-go/issues/2965 Yes, it's not ideal and needs improvement. :-(

For now, I guess vscode users will need to pass the build tags in both places. Something like

```
"gopls": {
    "buildFlags": ["-tags=iot,arduino,tinygo,gc.conservative,scheduler.coroutines"]
},
"go.buildTags": "iot,arduino,tinygo,gc.conservative,scheduler.coroutines"
```

go's `-tags` flag expects a comma-separated list.

bus710, 2020-04-13:
@hyangah,
Thanks a lot!

Now the code completion of Vscode works with this config (.vscode/settings.json):
```json
{
    "workbench.startupEditor": "newUntitledFile",
    "workbench.sideBar.location": "right",
    "editor.minimap.enabled": false,
    "workbench.colorTheme": "Dracula",
    "window.menuBarVisibility": "toggle",
    "go.formatTool": "goimports",
    "go.useLanguageServer": true,
    "git.enableSmartCommit": true,
    "git.autofetch": true,
    "[go]": {},
    "go.goroot": "/usr/local/go",
    "go.gopath": "/usr/local/tinygo:${env:HOME}/go",
    "go.toolsEnvVars": {
        "GOARCH": "arm",
        "GOOS": "linux",
    },
    "go.buildTags": "iot,arduino,tinygo,gc.conservative,scheduler.coroutines",
    "gopls": {
        "buildFlags": [
            "-tags=iot arduino tinygo gc.conservative scheduler.coroutines"
        ],
    }
}
```

One small hiccups is that Vscode warns **You are neither in a module nor in your GOPATH**. I guess this is because I deleted go.mod (but we shouldn't have this anyway to disable go module support).

hyangah, 2020-04-13:
@bus710 Awesome! I am slightly concerned about the "You are neither in a module nor in your GOPATH" error though. That's from `gopls` and that may indicate a bug. Do you mind filing a bug in golang.org/issues when you have a chance so we can continue debugging gopls from there? Many thanks!

* GOPATH: /usr/local/tinygo:${env:HOME}/go
* The location of your project directory relative to the above 'GOPATH'. 
* Gopls log (instruction: https://github.com/golang/tools/blob/master/gopls/doc/troubleshooting.md#vs-code)

sago35, 2020-08-31:
https://github.com/tinygo-org/tinygo-site/pull/107

sago35, 2020-08-31:
Here's a complete example that works with VSCode.
https://github.com/ecowden/hello-circuit-playground

deadprogram, 2020-10-04:
I think this issues was addressed in the most recent release, so now closing. Please reopen if needed.

closed 2020-10-04 by deadprogram

gadmin88, 2022-05-02:
Issue is still present, tested on  Windows 10 Pro/21H2/19044.1645, 

- JetBrains Goland 2022.1
- go1.18 windows/amd64
- tinygo version 0.23.0 windows/amd64 (using go version go1.18 and LLVM version 14.0.0)

Only solution is the way from @bgould in earlier [post ](https://github.com/tinygo-org/tinygo/issues/927#issuecomment-592221234)

[image]

artspb, 2022-05-02:
@gadmin88 Did you try [the dedicated plugin](https://plugins.jetbrains.com/plugin/16915-tinygo) for GoLand?

gadmin88, 2022-05-02:
@artspb yes, it is installed, with latest version 0.3.3

[image]

artspb, 2022-05-02:
@gadmin88 Indeed, one can see it on the first screenshot you've posted. Then you don't actually need to specify GOPATH manually. All configuration should be done under _Go | TinyGo_. Also, perhaps, it'd be better if you file an issue in [our tracker](https://youtrack.jetbrains.com/issues/GO) or contact us via _Help | Contact Support_ in the IDE, so we don't disturb anyone else here with our comments.

gadmin88, 2022-05-02:
@artspb nice that this **should be** the way, but it does **not** WORK, just tested 2 hours ago,
just for you here a screenshot with only the mentioned points:

**Settings:**
[image]
[image]
[image]
[image]
[image]

**Result:**
[image]

CrollUK, 2022-05-15:
@gadmin88 I disabled 'Go module integration' and I have it working. I was having the same experience as yourself.

charathram, 2022-07-13:
The "bug" here is that the TinyGo plugin for Goland doesn't find the cached GOROOT as is required by the TinyGo [documentation](https://tinygo.org/docs/guides/ide-integration/).

> The GOROOT value needed is the cached GOROOT given in the output (/home/user/.cache/tinygo/goroot-go1.14-f930d5b5f36579e8cbd1c139012b3d702281417fb6bdf67303c4697195b9ef1f-syscall in the example).

This seems like a bug with the TinyGo Goland plugin and not so much TinyGo itself.

nickfromXXII, 2022-08-02:
@charathram Could you please describe detailed steps to reproduce this issue? Do you face this problem on Windows? Which version of TinyGo plugin do you use? Do you still face the issue after IDE restart? It would be nice if you could submit an issue with detailed description on JetBrains tracker: https://youtrack.jetbrains.com/issues/GO.

charathram, 2022-08-03:
@nickfromXXII I've opened https://youtrack.jetbrains.com/issue/GO-13453/TinyGo-Plugin-does-not-detect-cached-GOROOT for this issue.