tinygo-org/tinygo #931: Concurrency only support on 32 bit microcontrollers?

evseevt, 2020-02-28:
Traget board : Arduino UNO.
Goal : want using Go concurrency on atmega328p.
Result : 2 led -> blink only one.

Code :
package main

import (
	"machine"
	"time"
)

func main() {
	go led1()
	led2()
}

func led1() {
	led := machine.Pin(2)
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 1000)

		led.High()
		time.Sleep(time.Millisecond * 1000)
	}
}

func led2() {
	led := machine.Pin(3)
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 420)

		led.High()
		time.Sleep(time.Millisecond * 420)
	}
}

jaddr2line, 2020-02-28:
Currently goroutines are not supported on AVR due to bugs in the LLVM backend.

aykevl, 2020-02-29:
Yeah, it's not yet supported. I started some work but never finished it.

jaddr2line, 2020-03-27:
PR #961 added support for goroutines. They are turned off by default though, as most AVR boards do not have all that much memory to work with.

closed 2020-03-27 by jaddr2line