tinygo-org/tinygo #940: AVR how to change CPU frequency, from 16MHz to 8MHz?

evseevt, 2020-03-04:
Traget board : Arduino UNO.
Goal : I want to change CPU frequency, from 16MHz to 8MHz. How can i do this?

In tinygo/src/machine/board_arduino.go 
```
// Return the current CPU frequency in hertz.
func CPUFrequency() uint32 {
	return 16000000
}
```
Result: I can use CPU only on 16MHz?

aykevl, 2020-03-04:
There is no support for changing the frequency at the moment.

Why do you want to run at a lower frequency? To be able to run at a lower voltage?

evseevt, 2020-03-05:
> There is no support for changing the frequency at the moment.
> 
> Why do you want to run at a lower frequency? To be able to run at a lower voltage?

Causes :
1) Using MCU without quartz resonator (enable internal clock).
**2) Needed for simulation MCU.**
3) Lower voltage case.

aykevl, 2020-03-05:
A lower frequency probably works just fine with TinyGo, but timers might be wrong. Are you using a different board than the Arduino Uno?

evseevt, 2020-03-06:
In combination, TinyGo and Proteus would give a big impetus to beginners and not only in the study of TinyGo. But now this bunch does not work and it is sad. I'm trying to set up a bunch, but for this I need to change the frequency in the .hex file. Maybe this will help.

closed 2020-03-10 by yevsya-git

aykevl, 2020-03-11:
I doubt changing the frequency is going to help. You can try it by manually patching the source file. In the end, if it works correctly on the hardware but not in an emulator then the emulator is wrong.

jkandasa, 2023-02-13:
@aykevl I see this is old issue. But still for me it looks valid
I am having different atmega328p boards with different frequency, 1Mhz, 8Mhz and 16 Mhz
I would like to change the default frequency. Is it possible with the following changes?
If yes I can create a PR. WDYT?
```go
var (
	cpuFrequency = uint32(16000000)
)

// Return the current CPU frequency in hertz.
func CPUFrequency() uint32 {
	return cpuFrequency
}

func SetCPUFrequency(freq uint32) {
	cpuFrequency = freq
}
```