tinygo-org/tinygo #948: SoftCore RISC-V Support for PicoRV32 or VexRiscv platforms.

trevor403, 2020-03-10:
It look like it should be possible to get TinyGo running on an FPGA softcore without too much effort.
I was reading this article earlier that describes using linker scripts for the HiFive1 and startup code for the fe310, which already exist in the TinyGo project.
https://blog.adacore.com/ada-on-fpgas-with-picorv32

The tricky part would be the machine package. There are several "machines" or SoCs that are currently in use in the RISC-V softcore world including
- PicoSoC https://github.com/cliffordwolf/picorv32/tree/master/picosoc
- Murax https://github.com/SpinalHDL/VexRiscv#murax-soc 
- MiSoC (aka LiteX SoC builder) https://github.com/m-labs/misoc
- Fomu's mix of PicoSoC and MiSoC https://github.com/im-tomu/foboot/tree/master/hw/rtl

So building out a machine package may be difficult. 

It seems the most commonly used pairs are VexRiscv+LiteX and PicoRV32+PicoSoC
There is a great simulator called [Renode](https://renode.readthedocs.io/en/latest/index.html
) that can emulate both of these configurations. 
Here is a great guide
https://risc-v-getting-started-guide.readthedocs.io/en/latest/zephyr-litex.html#running
I will try to hack on getting a program running. It would be great if FPGAs could be a supported platform for Golang

aykevl, 2020-03-11:
This sounds very interesting!

What the machine package does is give a common abstraction over specific hardware. In this case, the abstraction would probably cover one set of hardware peripherals.
Perhaps the best way to start is to implement support for one specific softcore and then see what kind of abstraction is needed to also support other softcores?

deadprogram, 2024-11-30:
This is in fact possible. Check out my PR https://github.com/tinygo-org/tinygo/pull/4631 where I am doing exactly this.

deadprogram, 2024-12-14:
That PR has now been merged into the `dev` branch. Let the FPGA games begin!

davecheney, 2026-10-24:
TinyGo now supports the Tillitis TKey, an FPGA-based PicoRV32 device, via https://github.com/tinygo-org/tinygo/pull/4631. This example blinks its LED:

```go
package main

import (
	"machine"
	"time"
)

func main() {
	machine.LED.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		machine.LED.High()
		time.Sleep(500 * time.Millisecond)
		machine.LED.Low()
		time.Sleep(500 * time.Millisecond)
	}
}
```

Save as `main.go`. With `tkey-runapp` installed and a TKey connected:

```sh
tinygo flash -target=tkey main.go
```

I verified that the sample compiles for `tkey`; I have not tested it on hardware. Support for other FPGA boards or VexRiscv-based SoCs should be tracked separately.

closed 2026-10-24 by davecheney