tinygo-org/tinygo #952: time.Now().Unix() returns 0 on Arduino Uno

artspb, 2020-03-12:
Here's the code.

```
package main

import (
	"time"
)

func main() {
	println(time.Now().Unix())
}
```

I do the following.

`tinygo flash -target=arduino -port=/dev/tty.usbmodem14301 main.go && screen -L /dev/tty.usbmodem14301 9600`

After closing `screen`, the following command `cat screenlog.0` prints `0`.

When I run code locally, it prints time more or less properly (see #68).

```
tinygo run main.go && date +%s
799780
1584031069
```

jaddr2line, 2020-03-12:
Most AVR chips (including the chip on the uno) lack real timing hardware. Right now, `time.Now()` is using the not-very-accurate timing mechanism that is used for `time.Sleep` on AVR. This timer is relative to startup time. In the future we may add an option to calibrate the timing, but the lack of an onboard clock means that this will not persist across restarts/power loss.

artspb, 2020-03-12:
In my case, a relative time would work. But it doesn't seem obvious to me how `time.Now()` works. I'd expect that from the following code that an LED would be always one. It never happens.

```
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	led.Low()
	for {
		if time.Now().Unix() != 0 {
			led.High()
		}
	}
}
```

But when I slightly modify the loop, it suddenly starts working.

```
if time.Now().Unix() != 0 {
	led.High()
	time.Sleep(time.Second)
	led.Low()
	time.Sleep(time.Second)
}
```

jaddr2line, 2020-03-13:
The way that AVR time works now is that it keeps track of the amount of time it has slept: https://github.com/tinygo-org/tinygo/blob/master/src/runtime/runtime_avr.go

aykevl, 2020-03-13:
> Also, for Arduino, it's [implemented](https://github.com/arduino/ArduinoCore-avr/blob/2f67c916f6ab6193c404eebe22efe901e0f9542d/cores/arduino/wiring.c) via timer interrupts. Could it make sense of doing the same in Go?

Yes, that should be done. But nobody has done that yet.

deadprogram, 2020-03-13:
We'd be delighted to receive a PR that added this feature. :)

cweagans, 2022-01-02:
I believe this is fixed by https://github.com/tinygo-org/tinygo/pull/2428

deadprogram, 2024-12-11:
Thanks @cweagans I just noticed your comment. Now closing.

closed 2024-12-11 by deadprogram