tinygo-org/tinygo #953: time.Now().String() doesn't compile for Arduino Uno

artspb, 2020-03-12:
I have the following code.

```
package main

import (
	"time"
)

func main() {
	_ = time.Now().String()
}
```

When I run the following command `tinygo flash -target=arduino -port=/dev/tty.usbmodem14301 main.go` I get this.

```
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main section `.text' will not fit in region `FLASH_TEXT'
/usr/local/opt/avr-binutils/bin/avr-ld: region `FLASH_TEXT' overflowed by 11019 bytes
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o: in function `LBB9_5':
main.go:(.text.(reflect.Value).Complex+0xb2): undefined reference to `__extendsfdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.(reflect.Value).Complex+0xd2): undefined reference to `__extendsfdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o: in function `LBB11_4':
main.go:(.text.(reflect.Value).Float+0x54): undefined reference to `__extendsfdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o: in function `LBB21_48':
main.go:(.text.runtime._panic+0x2c4): undefined reference to `__extendsfdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o: in function `LBB21_71':
main.go:(.text.runtime._panic+0x3fa): undefined reference to `__extendsfdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o:main.go:(.text.runtime._panic+0x406): more undefined references to `__extendsfdf2' follow
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o: in function `runtime.printfloat64':
main.go:(.text.runtime.printfloat64+0x48): undefined reference to `__unorddf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x6a): undefined reference to `__ledf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x80): undefined reference to `__adddf3'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0xaa): undefined reference to `__gtdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0xd2): undefined reference to `__ltdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x10a): undefined reference to `__gedf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x130): undefined reference to `__gtdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x158): undefined reference to `__ltdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x1e0): undefined reference to `__gtdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x206): undefined reference to `__ltdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x296): undefined reference to `__gedf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x2b6): undefined reference to `__subdf3'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x2e6): undefined reference to `__ltdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x2fe): undefined reference to `__divdf3'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x33c): undefined reference to `__gedf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x358): undefined reference to `__muldf3'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x3a0): undefined reference to `__adddf3'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x3bc): undefined reference to `__ltdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x3de): undefined reference to `__divdf3'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x418): undefined reference to `__divdf3'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x424): undefined reference to `__gedf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x486): undefined reference to `__fixdfsi'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x49e): undefined reference to `__floatsidf'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x4b4): undefined reference to `__subdf3'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.printfloat64+0x4ce): undefined reference to `__muldf3'
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o: in function `LBB43_15':
main.go:(.text.runtime.reflectValueEqual+0x170): undefined reference to `__eqdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: main.go:(.text.runtime.reflectValueEqual+0x196): undefined reference to `__eqdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o: in function `LBB43_29':
main.go:(.text.runtime.reflectValueEqual+0x31e): undefined reference to `__eqdf2'
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o: in function `LBB43_31':
main.go:(.text.runtime.reflectValueEqual+0x334): undefined reference to `__unorddf2'
collect2: error: ld returned 1 exit status
error: failed to link /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main: exit status 1
```

aykevl, 2020-03-12:
There are two important parts to this error:

```
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main section `.text' will not fit in region `FLASH_TEXT'
/usr/local/opt/avr-binutils/bin/avr-ld: region `FLASH_TEXT' overflowed by 11019 bytes
```

This means that the `.String()` method is way too big to fit on an Arduino.

```
/usr/local/opt/avr-binutils/bin/avr-ld: /var/folders/mk/r8k2br395m7djh4bwzktj2hc0000gn/T/tinygo948137028/main.o: in function `LBB9_5':
main.go:(.text.(reflect.Value).Complex+0xb2): undefined reference to `__extendsfdf2'
[... and many others like this ...]
```

This is caused by a lack of `float64` support in libgcc. It [might work with avr-gcc version 10](https://gcc.gnu.org/wiki/avr-gcc#Deviations_from_the_Standard), but then you would still have the problem that the flash is way too small for this.

In other words, I do not think this can be supported on a small chip such as the atmega328p.

artspb, 2020-03-12:
I see, thank you for the explanation.

closed 2020-03-12 by artspb