tinygo-org/tinygo #954: error: failed to flash // unable to locate UF2 device: PYBADGEBOOT

perryrivera, 2020-03-14:
**Distro**: Fedora 31
**tinygo version**: 0.11.0

$ **ls -al /dev/ttyACM0** 
crw-rw----. 1 root dialout 166, 0 Mar 14  2020 /dev/ttyACM0

$ **groups**
my_username wheel dialout docker

$ **tinygo flash -target=pybadge main.go**

error: failed to flash /tmp/tinygo_xxxxxx_/main.uf2: unable to locate UF2 device: PYBADGEBOOT

$ **id -a**
uid=_xxxx_(my_username) gid=_xxxx_(my_username) groups=_xxxx_(my_username),10(wheel),_xx_(dialout),971(docker) context=unconfined_(etc. etc.)

I've also tried:
$ **tinygo flash -port=/dev/ttyACM0 -target=pybadge main.go**

However...
I can easily copy UF2 files directly to the PYBADGEBOOT over USB with no issues.

How can I get tinygo flashing to work on Fedora 31?

aykevl, 2020-03-14:
What is the location of the PYBADGEBOOT directory? You can get it with the `mount` command. On Debian it's usually `/home/<username>/PYBADGEBOOT`.

aykevl, 2020-03-14:
As a temporary workaround, you can write it manually to the device. For example:

    tinygo build -o /home/$USER/PYBADGEBOOT/firmware.uf2 -target=pybadge main.go

deadprogram, 2020-03-14:
The location of this drive on Fedora is not the same as Debians. We ran into this exact problem recently at our workshop at SCaLE.

I think someone was going to submit a PR that added the correct path.

perryrivera, 2020-03-15:
> What is the location of the PYBADGEBOOT directory? You can get it with the `mount` command. On Debian it's usually `/home/<username>/PYBADGEBOOT`.

According to the Files app, the directory is:
/run/media/(my_username)/PYBADGEBOOT

perryrivera, 2020-03-15:
Thanks @aykevl !

> As a temporary workaround, you can write it manually to the device. For example:
> 
> ```
> tinygo build -o /media/$USER/PYBADGEBOOT/firmware.uf2 -target=pybadge main.go
> ```

This workaround appears to work (note the path below):
tinygo build -o /run/media/$USER/PYBADGEBOOT/firmware.uf2 -target=pybadge main.go

perryrivera, 2020-03-15:
> The location of this drive on Fedora is not the same as Debians. We ran into this exact problem recently at our workshop at SCaLE.
> 
> I think someone was going to submit a PR that added the correct path.

@deadprogram : Does the PR need to be submitted here or somewhere at Fedora? This looks like a tinygo building / flash issue. Does issue 954 serve as notice? :)

perryrivera, 2020-03-15:
Not sure if this related, but I get this new error trying to flash the name badge code:
$ **tinygo build -o /run/media/$USER/PYBADGEBOOT/firmware.uf2 -target=pybadge main.go**
# main.go
main.go:25:22: SPI1_MISO_PIN not declared by package machine

Changing the code from MISO to MIS0 doesn't appear to help....

deadprogram, 2020-03-16:
Regarding the path to the badge for flashing, that would be a PR here, please.

Regarding the SPI1 compile error, which program are you trying to build?

aykevl, 2020-03-18:
The relevant code is here: https://github.com/tinygo-org/tinygo/blob/master/main.go#L483
It locates the correct path simply by globbing at a well-known location (that unfortunately varies between distros).

Arsen6331, 2021-12-23:
Experiencing the same issue on Arch. The path for drives on Arch is `/run/media/<user>/<drive name>`.

Arsen6331, 2021-12-23:
Another workaround is creating a symlink, like so: `sudo ln -s /run/media/$USER /media/$USER`. After this, `tinygo flash` should just work.

aykevl, 2021-12-28:
Is there someone who would make a PR to fix this? The relevant code is here:

https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/main.go#L498-L499

It doesn't need any special knowledge, but is much easier done when you're actually running Fedora or Arch (which I don't).

Arsen6331, 2021-12-28:
I will try to make a PR. Should be ready soon if I don't encounter any issues.

Arsen6331, 2021-12-29:
I've just created a PR that seems to fix it. Only tested on Arch, not Fedora or Debian.

Arsen6331, 2021-12-29:
I accidentally made my previous PR against release. Sorry about that. Anyway, I've created a new PR and changed it to work with pre-1.16 versions. If someone could look at it, and test it on Arch and Fedora, that would be much appreciated. Thank you.

deadprogram, 2023-08-18:
I think that this has been resolved with https://github.com/tinygo-org/tinygo/pull/3816 so tagging to close on next release.

deadprogram, 2023-08-26:
Closing since this is part of release 0.29.0 thanks!

PR #2437 (2021-12-29): Check for /run/media on Linux for non-debian-based distros (fixed)
On non-debian-based distros, the directory for mounted drives is `/run/media` rather than `/media`

This PR uses `os.Stat()` to check if the `/run/media` directory exists on Linux, and changes the glob based on that.

This has only been tested on Arch, so further testing would be appreciated.

This is my second PR as I made the first one (#2436) against release accidentally. Sorry about that.

Fixes #954 and  #1740
closed 2023-08-26 by deadprogram