tinygo-org/tinygo #962: arduino-nano33-iot sleep times incorrect?

rck, 2020-03-17:
I wanted to port a library that handles [RC switches](https://github.com/sui77/rc-switch). The library works for me when I use it from the arduino IDE and with a given hardware setup. In the end this pulls up a data pin, waits, pulls it down, waits again, till all bits are sent. I never got it to work with tinygo.

To get the most basic test case, I eventually slightly modified the c++ lib to output `pin.High()/pin.Low()` when it would have pulled the pin high/low and for every sleep, I generated the according `time.Sleep()`. So basically this is a completely unrolled Go version of what the c++ code (successfully) does. Still, it does not work.

I can provide the whole code, but it would only really make sense if one uses the same RC switch with the same dip switch settings. Anyways, just that you get a feeling how this looks like:

```go
package main

import (
        "machine"
        "time"
)

const (
        led   = machine.LED
        rcPin = machine.D8
)

func main() {
        led.Configure(machine.PinConfig{Mode: machine.PinOutput})
        rcPin.Configure(machine.PinConfig{Mode: machine.PinOutput})

        turnOn(rcPin)
        for {
                led.High()
                time.Sleep(1 * time.Second)
                led.Low()
                time.Sleep(1 * time.Second)
        }
}

func turnOn(p machine.Pin) {
        //repeat #0
        p.High()
        time.Sleep(time.Duration(350 * time.Microsecond))
        p.Low()
        time.Sleep(time.Duration(1050 * time.Microsecond))
        p.High()
        time.Sleep(time.Duration(350 * time.Microsecond))
        p.Low()
        time.Sleep(time.Duration(1050 * time.Microsecond))
        p.High()
        time.Sleep(time.Duration(350 * time.Microsecond))
        p.Low()
        time.Sleep(time.Duration(1050 * time.Microsecond))
        p.High()
        time.Sleep(time.Duration(350 * time.Microsecond))
        p.Low()
        // many more such boring lines, ~1000 in total
}
```

The code actually repeats the protocol 10 times, so one switch to on is about 100 lines.

Any ideas whatsoever what I'm doing wrong or why this rather simple code does not work?

jaddr2line, 2020-03-17:
Suggestions for debugging:

1. Try running with a build of the dev branch. A lot has been changed, and this will eliminate the possibility that this has already been fixed.
2. Run this while connected to a power-only USB source (like a charger). USB interrupts may be interfering with your timing.
3. Turn the scheduler off. Use `-sched=none` on a build of the dev branch.

jaddr2line, 2020-03-17:
I do not think that unrolling this makes sense. 350 microseconds is about 17k clock cycles.

rck, 2020-03-18:
Hi @jaddr2line, thanks for your suggestions:

> 3. Turn the scheduler off. Use `-sched=none` on a build of the dev branch.

Now using `origin/dev` as of yesterday (`a8da6016723ed83f5148bc2908f990de7ef79ed8`). Guess you are talking about `-scheduler`, not `-sched`. But that is what makes the difference! Keeping it at the default, does not work. `-scheduler=none`, works perfectly fine. Actually none of the options (coroutines or tasks) works. It has to be "none".

So "case closed". If you you think it actually should work with a scheduler, feel free to point me to some branch and I will try it.

jaddr2line, 2020-03-18:
You only have one goroutine, so the scheduler should definitely be able to get somewhat close to your specified timing. I will run some tests on mine later today.

jaddr2line, 2020-03-18:
So I ran some tests using this code: https://gist.github.com/jaddr2line/298f002be6fdc80e1ebd3eeaf8ceb31d

I plotted the results:
https://docs.google.com/spreadsheets/d/1iLF2h2V7I9pl-INoWlVqni8-an_yIDse4KViiVlYIac/edit?usp=sharing

There is a lot more overhead in the timing setup than I would have thought. The overhead is 647 µs for both schedulers, and 261 µs when not using a scheduler.

deadprogram, 2020-03-26:
Is this issue this a problem with the latest dev branch?

rck, 2020-03-26:
I only tried the microcontroller side of things and there (c4fd19be99c2695f25ac59057adef7a44ad61ad9) it did not change anything. `-scheduler=none` works, without that it does not.

deadprogram, 2020-05-28:
I think that PR #1127 should address a lot of timing issues, possibly including this one.

deadprogram, 2021-09-18:
Closing due to inactivity. Please reopen if needed. Thank you!

closed 2021-09-18 by deadprogram