tinygo-org/tinygo #980: 3DS Support

superloach, 2020-03-23:
This will probably be a difficult endeavor, but I'd like to propose support for the Nintendo 3DS as a build target.

Things I know:
 - Architecture: ARM11 MPCore
 - Display: DMP PICA200

Things I've tried:
 - [justMaku/Swift3DS](https://github.com/justMaku/Swift3DS/blob/master/Makefile) uses the `arm-none-eabihf` LLVM target, which doesn't have runtime code in TinyGo

Things to do:
 - Implement runtime for `arm-none-eabihf`
 - Write library for platform-specific stuff (likely `package n3ds`)

jaddr2line, 2020-03-23:
So there are a few things you need to implement in the runtime in order to add a buildable target:
1. timing code - this means that you need to provide an implementation of `runtime.ticks` (checking a monotonic timer), and `runtime.sleepTicks` (sleeping until a target time)
2. console output - you need to implement `runtime.putchar`, which will be used by the `print`/`println` builtins (I am not entirely sure where this output should go, as programs are likely to be using the screen for something else)
3. an entry point - for nds it looks like this should be an extern symbol called `main`, and should do any necessary pre-initialization before calling `runtime.run` (which will initialize all packages and then start the Go program)
4. a heap - before initializing the runtime, you need to provide a contiguous chunk of memory which can be used for allocations; it is not immediately obvious where to get it from looking at swit3ds
5. a target file - specifies build tags/llvm target/etc.; for example: [wasm target file](https://github.com/tinygo-org/tinygo/blob/e2aa3789c303a01c65ebabd668df7a7e65ccacde/targets/wasm.json)

Some low-level bindings can go in the `machine` package; but high level drivers (e.g. the display drivers) should be put somewhere else (e.g. https://github.com/tinygo-org/drivers or maybe a seperate 3DS repo).

aykevl, 2020-03-23:
You can look at the GBA PR here for inspiration: https://github.com/tinygo-org/tinygo/pull/467 (but note that some things have changed since then such as the `run` function).
Some things (such as timing and serial output) are not really necessary, especially at the beginning.

I would recommend looking for an absolute minimal example of something written in C, with no external libraries. That can given you an idea what is involved in building code for the 3DS.

> [justMaku/Swift3DS](https://github.com/justMaku/Swift3DS/blob/master/Makefile) uses the `arm-none-eabihf` LLVM target, which doesn't have runtime code in TinyGo

It's not the architecture that needs to be supported but the specific chip (the architecture is already supported by LLVM and only really describes the instructions, not all the memory mapped peripherals and startup sequence). The 3DS is based on the ARM11 but that's just one of the processor cores in the chip. I think this chip is purpose-built for the 3DS.