tinygo-org/tinygo #984: panic in interface lowering pass

bgould, 2020-03-24:
The following program results in a panic when compiling:
```
tinygo build -o test main.go
```

```go
package main

import (
	"machine"
	"time"

	"github.com/Omegaice/go-xmodem/xmodem"
)

func main() {
	bytes, err := xmodem.ModemReceive(machine.UART0)
	for {
		time.Sleep(30 * time.Second)
		if err != nil {
			println("Error: ", err)
		} else {
			println("content: ", string(bytes))
		}
	}
}
```
This is the stack trace for the panic:
```
panic: expected exactly one call use of runtime.interfaceMethod

goroutine 1 [running]:
github.com/tinygo-org/tinygo/transform.(*lowerInterfacesPass).run(0xc0081bd320)
        /tinygo/src/github.com/tinygo-org/tinygo/transform/interface-lowering.go:318 +0x1f86
github.com/tinygo-org/tinygo/transform.LowerInterfaces(0x2ad3230)
        /tinygo/src/github.com/tinygo-org/tinygo/transform/interface-lowering.go:160 +0x17d
github.com/tinygo-org/tinygo/transform.Optimize(0x2ad3230, 0xc0001c8030, 0x2, 0x2, 0x5, 0x0, 0x0, 0x0)
        /tinygo/src/github.com/tinygo-org/tinygo/transform/optimizer.go:78 +0x3da
github.com/tinygo-org/tinygo/builder.Build(0x7fff6e383254, 0x7, 0x7fff6e38324f, 0x4, 0xc0001c8030, 0xc0081bdca8, 0x0, 0x0)
        /tinygo/src/github.com/tinygo-org/tinygo/builder/build.go:84 +0x2a8f
main.Build(0x7fff6e383254, 0x7, 0x7fff6e38324f, 0x4, 0xc0001b8000, 0x0, 0xe)
        /tinygo/src/github.com/tinygo-org/tinygo/main.go:96 +0xc5
main.main()
        /tinygo/src/github.com/tinygo-org/tinygo/main.go:787 +0x1912
```
When compiled with https://github.com/tinygo-org/tinygo/pull/981 the following message is the result:
```
../../../../Omegaice/go-xmodem/xmodem/xmodem.go: internal error: expected exactly one call use of runtime.interfaceMethod
```
FWIW commenting out these lines seems to eliminate the panic: https://github.com/Omegaice/go-xmodem/blob/master/xmodem/xmodem.go#L220-L227

aykevl, 2020-03-24:
That's an interesting bug, and it appears to be introduced by a LLVM optimization pass that does something I didn't expect. Here is a fix: https://github.com/tinygo-org/tinygo/pull/986

jaddr2line, 2020-04-13:
Fix has been merged and released.

closed 2020-04-13 by jaddr2line