tinygo-org/tinygo #993: Large shifts have UB

jaddr2line, 2020-03-26:
Currently, tinygo does not process large right shifts according to spec, due to UB inside LLVM.

In the [Go spec](https://golang.org/ref/spec#Arithmetic_operators):
```
There is no upper limit on the shift count. Shifts behave as if the left operand is shifted n times by 1 for a shift count of n. As a result, x << 1 is the same as x*2 and x >> 1 is the same as x/2 but truncated towards negative infinity.
```

In LLVM's language specification for [`lshr`](https://llvm.org/docs/LangRef.html#lshr-instruction):
```
This instruction always performs a logical shift right operation. The most significant bits of the result will be filled with zero bits after the shift. If op2 is (statically or dynamically) equal to or larger than the number of bits in op1, this instruction returns a poison value.
```

There is a discrepancy here, and it breaks some go code. Here is an example which works on 64-bit systems but crashes due to UB optimizations on 32/16-bit systems:
```
package main

func val(v uintptr) uintptr {
	return v
}

func main() {
	if val(2)>>32 == 0 {
		panic("this is supposed to panic")
	}
	println("this should not happen")
}
```

jaddr2line, 2020-03-26:
I would like to propose the following IR for right shifts (assuming uint32 here; `a >> b`):
```
%shift.val = lshr i32 %a, %b
%shift.overflow = icmp ult i32 %b, 32
%shift.result = select i1 %shift.overflow, i32 0, i32 %shift.val
```

This will have a negative performance impact if the bounds of `b` are not known at compile time, however I do not see any viable alternative.

If the shift is signed, we can also add an assert for that.

I suspect that this will not have a huge impact, as most shifts that I have seen are by constants, and thus do not require any branches.

aykevl, 2020-03-26:
Yes, I think that's the best way to work around it. If `%b` is 32 or larger, `%shift.val` will be a poison value (not directly UB). The `select` will not trigger the poison because it will pick 0 instead on overflow.

aykevl, 2020-03-26:
If this causes performance problems, you can probably work around it in code with masking (on ARM with `& 255` and on x86 with `& <width-1>`).

jaddr2line, 2020-03-28:
This has been fixed by #999.

closed 2020-03-28 by jaddr2line