tinygo-org/tinygo #1003: interface conversion: ssa.Member is nil, not *ssa.Function (easy repro)

rsms, 2020-03-28:
Repro: Compile main.go, provided below, with:

```sh
$ tinygo build -o main.wasm -target wasm main.go`
panic: interface conversion: ssa.Member is nil, not *ssa.Function

goroutine 1 [running]:
github.com/tinygo-org/tinygo/ir.(*Program).SimpleDCE(0xc002953d80)
	/Users/distiller/project/ir/passes.go:69 +0xb61
github.com/tinygo-org/tinygo/compiler.(*Compiler).Compile(0xc0000b1930, 0x7ffeefbff940, 0x7, 0xc0000b1930, 0x0, 0x0)
	/Users/distiller/project/compiler/compiler.go:277 +0x9ad
github.com/tinygo-org/tinygo/builder.Build(0x7ffeefbff940, 0x7, 0x7ffeefbff920, 0x12, 0xc0001ba330, 0xc002419ca8, 0x0, 0x0)
	/Users/distiller/project/builder/build.go:35 +0xab
main.Build(0x7ffeefbff940, 0x7, 0x7ffeefbff920, 0x12, 0xc0001c2000, 0x0, 0x12)
	/Users/distiller/project/main.go:85 +0xc5
main.main()
	/Users/distiller/project/main.go:767 +0x1903
```

main.go
```go
package main
import "fmt"
func logf(format string, v... interface{}) {
  fmt.Printf(format, v...)
}
```

Environment:
- macOS 10.14
    - `uname -v` → `Darwin Kernel Version 18.7.0: Thu Jan 23 06:52:12 PST 2020; root:xnu-4903.278.25~1/RELEASE_X86_64`
- tinygo 0.12.0 (from Homebrew)

jaddr2line, 2020-03-28:
This is actually a bug in your code - it is missing a `main` function.

I do agree however that we need a better error message.

jaddr2line, 2020-03-28:
I put together a quick fix in #1004, which should provide an understandable error message.

aykevl, 2020-03-28:
The fix to the panic has been merged in #1004.

I think there is nothing left to do here, so closing. Thanks again @jaddr2line for the quick fix!

closed 2020-03-28 by aykevl