tinygo-org/tinygo #1009: machine: Arduino Nano33 IoT crashing on mqtt publish after some period of time

deadprogram, 2020-03-29:
The latest `dev` branch has some issues when running the https://github.com/tinygo-org/drivers/tree/master/examples/wifinina/mqttsub example.

After some period of time, somewhere between 100 and 1500 messages sent/received, the program on the Arduino Nano33 IoT crashes, giving one of the following outputs:

```
panic: runtime error: out of memory
```

```
fatal error: HardFault with sp=0x200007b4 pc=0xf7f8fbbd
````

```
panic: runtime error: goroutine stack overflow
```

```
panic: runtime error: nil pointer dereference
```

Note that it is one of the above when it crashes, not all of them.

aykevl, 2020-03-29:
This looks like a memory corruption. Does it also happen with `-scheduler=coroutines`?

aykevl, 2020-03-29:
Also, it would help a lot if you could bisect this issue to the offending commit (assuming that the master branch works).

deadprogram, 2020-03-29:
It runs without errors when I use

```
tinygo flash -target arduino-nano33 -scheduler=coroutines ./examples/wifinina/mqttsub/
```

I will bisect to try to find the commit that breaks things.

deadprogram, 2020-03-29:
It appears this is the answer:

```
# first bad commit: [6a50f25a48385fc78397b6f0f1c83650130d8d16] refactor coroutine lowering and tasks
```

aykevl, 2020-03-29:
Okay, great. That's at least something. @jaddr2line can you perhaps take a look?

jaddr2line, 2020-03-29:
I suspect that this is due to a small change in stack usage causing an overflow. Can you try raising the stack size a bit? Those numbers sound like potential garbage collection times.

jaddr2line, 2020-03-30:
Also the only relevant change I can think of from that commit is that the task data is no longer on top of the stack. It may have overflowed before, but is now corrupting something other than the task data.

aykevl, 2020-03-30:
Here is the stack size you can try raising (for example to 2048):

https://github.com/tinygo-org/tinygo/blob/57320c09222560dd3656d3c4bb05a16c1849e9a5/src/internal/task/task_stack_cortexm.go#L7

deadprogram, 2020-03-30:
Raising the stack size to 2048 had no effect.

jaddr2line, 2020-03-30:
There appears to be an issue with GC tracking right now. This patch forces the `conservative` GC to reclaim memory earlier for debugging purposes:
```
diff --git a/src/runtime/gc_conservative.go b/src/runtime/gc_conservative.go
index 319e68b..5fb87b3 100644
--- a/src/runtime/gc_conservative.go
+++ b/src/runtime/gc_conservative.go
@@ -219,8 +219,11 @@ func alloc(size uintptr) unsafe.Pointer {
 
        neededBlocks := (size + (bytesPerBlock - 1)) / bytesPerBlock
 
+       GC()
+
        // Continue looping until a run of free blocks has been found that fits the
        // requested size.
+       nextAlloc = 0
        index := nextAlloc
        numFreeBlocks := uintptr(0)
        heapScanCount := uint8(0)
```

This causes several compiler tests to fail, and some have errors similar to what have been discovered here. It also triggers the bug discovered at https://github.com/tinygo-org/tinygo/pull/1012#issuecomment-606183993.

jaddr2line, 2020-03-31:
With stack slots, it appears that we are not properly tracking phi nodes:
```
package main

import "math/rand"

func gen(n int) *int {
	return &n
}

func check(ptr *int, n int) {
	if *ptr != n {
		panic("corrupted memory")
	}
}

func main() {
	a, b := 0, 1
	aptr, bptr := gen(a), gen(b)
	for i := 0; i < 1000; i++ {
		println(i)

		if rand.Intn(2) == 0 {
			a, b = b, a
			aptr, bptr = bptr, aptr
		}

		a = rand.Int()
		aptr = gen(a)

		check(aptr, a)
		check(bptr, b)
	}
}
```

However, this does not explain corruption on bare-metal platforms.

jaddr2line, 2020-04-01:
I have confirmed that some of the problems on baremetal are caused by references being stored in callee-saved registers. We do not currently scan these, and thus the allocations are freed early.

aykevl, 2020-04-02:
It took me a while to understand but now I see. This is the `main` function (`opt=1 -no-debug`):

```llvm
define dso_local void @"tmp/phi.go.main"(i8* nocapture readnone %context, i8* nocapture readnone %parentHandle) unnamed_addr section ".text.tmp/phi.go.main" {
entry:
  %gc.stackobject = alloca { %runtime.stackChainObject*, i64, i8*, i8*, i8* }
  store { %runtime.stackChainObject*, i64, i8*, i8*, i8* } { %runtime.stackChainObject* null, i64 3, i8* null, i8* null, i8* null }, { %runtime.stackChainObject*, i64, i8*, i8*, i8* }* %gc.stackobject
  %0 = load %runtime.stackChainObject*, %runtime.stackChainObject** @runtime.stackChainStart
  %1 = getelementptr { %runtime.stackChainObject*, i64, i8*, i8*, i8* }, { %runtime.stackChainObject*, i64, i8*, i8*, i8* }* %gc.stackobject, i32 0, i32 0
  store %runtime.stackChainObject* %0, %runtime.stackChainObject** %1
  %2 = bitcast { %runtime.stackChainObject*, i64, i8*, i8*, i8* }* %gc.stackobject to %runtime.stackChainObject*
  store %runtime.stackChainObject* %2, %runtime.stackChainObject** @runtime.stackChainStart
  %3 = tail call fastcc i64* @"tmp/phi.go.gen"(i64 0)
  %4 = bitcast i64* %3 to i8*
  %5 = getelementptr { %runtime.stackChainObject*, i64, i8*, i8*, i8* }, { %runtime.stackChainObject*, i64, i8*, i8*, i8* }* %gc.stackobject, i32 0, i32 2
  store i8* %4, i8** %5
  %6 = tail call fastcc i64* @"tmp/phi.go.gen"(i64 1)
  %7 = bitcast i64* %6 to i8*
  %8 = getelementptr { %runtime.stackChainObject*, i64, i8*, i8*, i8* }, { %runtime.stackChainObject*, i64, i8*, i8*, i8* }* %gc.stackobject, i32 0, i32 3
  store i8* %7, i8** %8
  br label %for.body

for.body:                                         ; preds = %for.body, %entry
  %9 = phi i64 [ 0, %entry ], [ %20, %for.body ]
  %10 = phi i64* [ %6, %entry ], [ %spec.select1, %for.body ]
  %11 = phi i64* [ %3, %entry ], [ %17, %for.body ]
  %12 = phi i64 [ 1, %entry ], [ %spec.select, %for.body ]
  %13 = phi i64 [ 0, %entry ], [ %16, %for.body ]
  tail call fastcc void @runtime.printint64(i64 %9)
  tail call fastcc void @runtime.printnl()
  %14 = tail call fastcc i64 @"math/rand.Intn"()
  %15 = icmp eq i64 %14, 0
  %spec.select = select i1 %15, i64 %13, i64 %12
  %spec.select1 = select i1 %15, i64* %11, i64* %10
  %16 = tail call fastcc i64 @"math/rand.Int"()
  %17 = tail call fastcc i64* @"tmp/phi.go.gen"(i64 %16)
  %18 = bitcast i64* %17 to i8*
  %19 = getelementptr { %runtime.stackChainObject*, i64, i8*, i8*, i8* }, { %runtime.stackChainObject*, i64, i8*, i8*, i8* }* %gc.stackobject, i32 0, i32 4
  store i8* %18, i8** %19
  tail call fastcc void @"tmp/phi.go.check"(i64* nonnull %17, i64 %16)
  tail call fastcc void @"tmp/phi.go.check"(i64* %spec.select1, i64 %spec.select)
  %20 = add nuw nsw i64 %9, 1
  %exitcond = icmp eq i64 %20, 1000
  br i1 %exitcond, label %for.done, label %for.body

for.done:                                         ; preds = %for.body
  store %runtime.stackChainObject* %0, %runtime.stackChainObject** @runtime.stackChainStart
  ret void
}
```

I checked that the value pointed to by `bptr` is somehow corrupted. Inspecting the IR, I can also see why: there is a loop in the phi nodes:

```llvm
  %10 = phi i64* [ %6, %entry ], [ %spec.select1, %for.body ]
  %spec.select1 = select i1 %15, i64* %11, i64* %10
```

Because this is a loop, the value can stay in `%10` / `%spec.select1` without being tracked.

jaddr2line, 2020-04-02:
#1018 should clear up the 3rd (and hopefully last) gc tracking bug

jaddr2line, 2020-04-02:
#1017 fixed the issue of certain tests failing on WASM when the GC was forced to re-collect before every allocations

jaddr2line, 2020-04-02:
@aykevl It is also entirely possible to get similar results without a loop in the phi nodes, but I have been unable to find an example in which LLVM does not optimize it out.

jaddr2line, 2020-04-08:
Are we attempting to get the fixes all in for v0.13?

deadprogram, 2020-04-08:
Yes, we have to, IMO.

jaddr2line, 2020-04-10:
All of the related PRs have been merged.

closed 2020-04-10 by jaddr2line