tinygo-org/tinygo #1015: BPF?

majgis, 2020-04-01:
I watched the video [Netflix talks about Extended BPF: A new software type](https://www.youtube.com/watch?v=7pmXdG8-7WU)
where Brendan Gregg mentions that new languages being designed to target BPF.
It made me wonder if tinygo would be a good candidate to target BPF.  I see LLVM can target BPF.  The first question is, conceptually, could you see it as part of this project, or would it be better to have a separate project that solely targeted BPF?

aykevl, 2020-04-01:
I know very little about BPF, apart from that it is fairly limited. It might be able to port TinyGo to it, but there will be limitations. How severe those limitations are mainly depend on whether it is possible to have a heap.

TinyGo has many targets (some of which are highly experimental) so if you get it to work on BPF it would be fine to include it as part of this project.

majgis, 2020-04-01:
Ok, I see this project:

majgis, 2020-04-01:
I see this project:
https://github.com/iovisor/gobpf
I need to dig deeper to understand motivations for one over the other.
I think targeting BPF is not necessary a resource constrained environment, but  concerns would be around introducing latency in certain cases. I'll continue to research but wanted to throw the idea on the table.  I'll close this issue, unless I find something compelling or someone else comments

aykevl, 2020-04-01:
I believe gobpf is a very different project. It seems to concern itself mainly with loading, running (and compiling?) BPF programs whereas TinyGo is a compiler that could potentially add support for eBPF as a backend (but running those eBPF programs would be out of scope except for testing).

majgis, 2020-04-02:
Ok, got it.
This document helped me understand a little better:
https://www.phoronix.com/scan.php?page=news_item&px=LLVM-BPF-VM-Backend-Lands
I think my initial intuition that BPF is a tiny en

closed 2020-04-02 by majgis