tinygo-org/tinygo #1037: time.NewTicker / time.NewTimer causes a runtime error:

ryskiwt, 2020-04-08:
Using time.NewTicker causes a runtime error:

```go
package main

import (
	"time"
)

func main() {
	ticker := time.NewTicker(time.Second)
	defer ticker.Stop()

	<-ticker.C

	neverDone := make(chan struct{})
	<-neverDone
}
```

Error:

```
Uncaught (in promise) LinkError: WebAssembly.instantiate(): Import #12 module="env" function="time.startTimer" error: function import requires a callable
```

Compiled using the 0.12.0 docker image:

```
docker run ... \
    -e GOOS=js -e GOARCH=wasm \
    tinygo/tinygo:0.12.0 tinygo build -target wasm -o hoge.wasm .
```

jaddr2line, 2020-04-08:
TinyGo does not implement the `runtimeTimer` system which is used to implement `time.NewTicker` and `time.AfterFunc`. We either need to somehow implement it (not currently possible due to how `//go:linkname` is implemented), or replace parts of the `time` package.

ryskiwt, 2020-04-08:
Thanks for reply. I got it.
Is there any timeline or roadmap for implementing `go:linkname` ?

jaddr2line, 2020-04-08:
So `//go:linkname` has been implemented, but in TinyGo it preserves type information. Since `time.runtimeTimer` is an unexported type, another package cannot linkname `time.startTimer`, because the types would be different and cause a compiler error.

jaddr2line, 2020-04-08:
For now, if possible, use `time.Sleep` or loops. I do not know when we are going to be able to efficiently implement this.

jaddr2line, 2020-04-26:
As a workaround, I wonder whether we could do something like `//go:linkname` but with types.

aykevl, 2020-04-29:
Maybe we need to support `//go:linkname` with slightly-mismatched types after all. Yes, we could allow custom type names for named types, but doing something like that would still be quite invasive. I still don't like the `//go:extern` trick for globals, it requires parsing the AST. I would prefer to avoid one more dependency on having the AST available while building the LLVM IR.

Slightly-mismatched types would be much less of a problem if [opaque pointer types](https://groups.google.com/forum/#!topic/llvm-dev/Dw_DYSXGFto) were completed in LLVM. Unfortunately, it looks like it'll still take quite a while (if it is ever finished). Until then, maybe the simplest solution is the following: if a function to call has a changed linkname (with `//go:linkname`), then check whether the types are the same and if not bitcast the function pointer. I think that would be the least invasive change to the compiler. Perhaps it would be easier after #1008.

aykevl, 2020-05-29:
#1008 is unrelated.

I had another idea how `//go:linkname` could be fixed: every pointer to a named struct could be cast to `*i8` or similar during the call. I don't expect that to affect optimizations, but it should be checked to be sure. It does add a bit of noise to the IR, however.

deadprogram, 2020-09-12:
I still really would like to be able to write code like 

```
case <-time.NewTimer(3 * time.Second).C:
```

aykevl, 2020-09-13:
Yes, and we will need this to support the stdlib testing package (which makes use of these timers).

aykevl, 2020-09-21:
I have an implementation for timers and tickers, will clean up the patch and send a PR soon.

aykevl, 2020-09-22:
Here is the PR: #1402

Zyko0, 2021-11-11:
Any update on this ?

aykevl, 2021-11-11:
I think I can take a look after #2011 has been merged.

Zyko0, 2021-11-21:
Any idea about when this could happen now #2011 has been merged ?
Thanks !

deadprogram, 2022-09-30:
Released as part of `v0.26.0`

closed 2022-09-30 by deadprogram