tinygo-org/tinygo #1045: Inject build-time variables with wasm target

patmood, 2020-04-10:
With go you can inject variables using ldflags like this:

`go build -o ./main.wasm -ldflags "-X main.Version=$(VERSION)" ./main.go`

Attempting this with tinygo:

`tinygo build -o ./main.wasm -target wasm -ldflags "-X main.Version=$(VERSION)"  ./main.go`

Gives:
```
tinygo:wasm-ld: error: unknown argument: -X
tinygo:wasm-ld: error: cannot open main.Version=1.0.0: No such file or directory
error: failed to link /var/folders/jc/3lr9ltt96k90h_mg_xq6pvbw0000gn/T/tinygo833831446/main: failed to link using built-in wasm-ld
```

Is there an equivalent way to achieve this in tinygo?

aykevl, 2020-04-11:
Right now, unfortunately not. However I think functionality like this would be useful in the long term (for example, to set a WiFi password at compile time when you're building firmware for a personal IoT-like device).

deadprogram, 2020-09-12:
@aykevl I really need this feature for something. Any thoughts on implementation that we could work on?

deadprogram, 2021-04-09:
This is now in the `dev` branch and will be included in the next release.

commit 82cc9cc1 (2021-04-06): main: implement -ldflags="-X ..."

This commit implements replacing some global variables with a different
value, if the global variable has no initializer. For example, if you
have:

    package main

    var version string

you can replace the value with -ldflags="-X main.version=0.2".

Right now it only works for uninitialized globals. The Go tooling also
supports initialized globals (var version = "<undefined>") but that is a
bit hard to combine with how initialized globals are currently
implemented.

The current implementation still allows caching package IR files while
making sure the va [... cut]
commit 92d77edb (2021-04-08): main: implement -ldflags="-X ..."

This commit implements replacing some global variables with a different
value, if the global variable has no initializer. For example, if you
have:

    package main

    var version string

you can replace the value with -ldflags="-X main.version=0.2".

Right now it only works for uninitialized globals. The Go tooling also
supports initialized globals (var version = "<undefined>") but that is a
bit hard to combine with how initialized globals are currently
implemented.

The current implementation still allows caching package IR files while
making sure the va [... cut]
PR #1775 (2021-04-09): Inject build time variables
This PR implements replacing some global variables with a different value, if the global variable has no initializer. For example, if you have:

```go
package main

var version string
```

you can replace the value with `-ldflags="-X main.version=0.2"`.

Right now it only works for uninitialized globals. The Go tooling also supports initialized globals (`var version = "<undefined>"`) but that is a bit hard to combine with how initialized globals are currently implemented.

The current implementation still allows caching package IR files while making sure the values  [... cut]
commit 33f76d1c (2021-04-09): main: implement -ldflags="-X ..."

This commit implements replacing some global variables with a different
value, if the global variable has no initializer. For example, if you
have:

    package main

    var version string

you can replace the value with -ldflags="-X main.version=0.2".

Right now it only works for uninitialized globals. The Go tooling also
supports initialized globals (var version = "<undefined>") but that is a
bit hard to combine with how initialized globals are currently
implemented.

The current implementation still allows caching package IR files while
making sure the va [... cut]
closed 2021-05-12 by aykevl