tinygo-org/tinygo #1047: nil pointer dereference panic when using ctx.Done() in select

bgould, 2020-04-12:
The program below works in standard Go but results in a nil pointer dereference in TinyGo using the dev branch:

```
# tinygo info
LLVM triple:       x86_64--linux
GOOS:              linux
GOARCH:            amd64
build tags:        linux amd64 tinygo gc.extalloc scheduler.coroutines
garbage collector: extalloc
scheduler:         coroutines

# cat main.go 
package main

import (
        "context"
        "time"
)

func main() {
        numbers := make(chan int, 1)
        go func() {
                n := 0
                for {
                        numbers <- n
                        n++
                        time.Sleep(1 * time.Second)
                }
        }()
        ctx := context.Background()
        for {
                select {
                case n := <-numbers:
                        println(n)
                case <-ctx.Done():
                        println("done")
                        return
                }
        }
}

# tinygo run .
0
panic: runtime error: nil pointer dereference
error: failed to run compiled binary /tmp/tinygo345449133/main: signal: aborted

```

jaddr2line, 2020-04-12:
I appear to have forgotten a nil check on the channel here: https://github.com/tinygo-org/tinygo/blob/3862d6e8a2fd8d2fc69ce59f43fb7eeb787e2538/src/runtime/chan.go#L515

jaddr2line, 2020-04-12:
This should be fixed by #1050

PR #1050 (2020-04-13): runtime (chan): fix blocking select on a nil channel
Previously, a blocking select on a nil channel would result in a nil panic inside the channel runtime code. This change fixes the nil checks so that the select works as intended.

Fixes: #1047
closed 2020-04-13 by jaddr2line