tinygo-org/tinygo #1049: Example at top of README doesn't compile

alexflint, 2020-04-12:
When building the example at the top of the readme, I get `LED not declared by package machine`

```go
$ tinygo version
tinygo version 0.12.0 linux/amd64 (using go version go1.14)

$ tinygo build -o tinygo-tutorial
# .
blink.go:9:20: LED not declared by package machine
```

jaddr2line, 2020-04-12:
You are compiling to a target (desktop linux) that doesnt have an LED. You need to use a target with an LED (e.g. an arduino)

jbpratt, 2020-11-23:
I think this can be closed, the docs seem clear enough: `Here is an example program that blinks the built-in LED when run directly on any supported board with onboard LED:`, `The above program can be compiled and run without modification on an Arduino Uno, an Adafruit ItsyBitsy M0, or any of the supported boards that have a built-in LED, just by setting the correct TinyGo compiler target. For example, this compiles and flashes an Arduino Uno:` and a build command right below.

deadprogram, 2020-11-24:
Sounds good, now closing.

closed 2020-11-24 by deadprogram