tinygo-org/tinygo #1061: Use for scripting?

assyrianic, 2020-04-19:
Hello TinyGo devs. I am wondering if TinyGo can be used as an embeddable scripting language in a C or C++ application?

bgould, 2020-04-19:
Probably not.  TinyGo is an alternate compiler for the standard Go language, which is strongly typed and generally statically compiled.  It is not an interpreter support just-in-time compilation of scripts, which I imagine is probably the type of functionality you are looking for.

assyrianic, 2020-04-20:
oh no, I don't need JIT compilation of scripts. I'm looking for an avenue of which to write compiled plugins for a game engine written in C++. The current scripting platforms for the engine is a statically typed, compiled scripting language that has only one data type (32-bit signed int) and the other choice is Python or Lua and I dislike both.

I'm interested in TinyGo solely due to the fact that it is strongly typed and statically compiled. This helps facilitate and scale larger game projects/mods and stability.

aykevl, 2020-04-26:
So basically you want to compile "scripts" written in Go using TinyGo and link that with the rest of the game? Is that correct?

I think it would be possible. What might make it more difficult is the assumption that TinyGo produces standalone binaries. You can also let it output object files but they still contain a `main` function. Additionally, you can't just call into the produced code - you first have to initialize the (small) runtime for GC to work correctly. But TinyGo does support the `//export` pragma which should export any function to C (similar to CGo) which you can then call from C.

assyrianic, 2020-04-26:
@aykevl  yes, the current (C++) game engine I work with has no default scripting platform (more flexible I guess) so developers and enthusiasts can can setup their own scripting platforms and create communities around said platform.

If I had to choose a language with garbage collection, I'd prefer Golang than say Python or Lua, Golang, especially from TinyGo, has the best mix of performance + simplicity in my books.

assyrianic, 2023-09-22:
Hi, sorry to bring this back from 3 years but I wanted to ask if TinyGo has or can support non-LLVM backends. I am still considering TinyGo for "external" scripting but all I would require is a custom backend that I can have output specific bytecode for a VM interpreter.

aykevl, 2023-09-22:
> Hi, sorry to bring this back from 3 years but I wanted to ask if TinyGo has or can support non-LLVM backends.

Unfortunately not. That would mean a near-complete rewrite of the compiler.

deadprogram, 2024-12-04:
I would suggest look into WASM for this purpose. In any case, now closing since question was answered.

closed 2024-12-04 by deadprogram