tinygo-org/tinygo #1062: Better error reporting of unsupported flags

aykevl, 2020-04-19:
There are a few flags that currently don't have proper error checking. The ones I found are `-gc`, `-scheduler` and `-size`. Using the wrong value will lead to no or unexpected errors.

I think the best way to handle it is to add a new method to `compileopts.Config` that checks whether there are any faulty flags, which is then called from the main function.

trevor403, 2020-04-20:
I can take this issue. I'll post a PR tomorrow.

cebernardi, 2020-04-26:
@trevor403 I just saw your comment here, I was "assigned" this issue on slack 19th of April..
I'm sorry we overlapped

trevor403, 2020-04-28:
That is totally okay! :smile:

jaddr2line, 2020-06-27:
Has this issue been sufficiently resolved?

aykevl, 2020-07-03:
Yes, it will be included in the next release (see the label).

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram