tinygo-org/tinygo #1063: Release Static Builds?

prologic, 2020-04-20:
Can you please consider adding static builds to the Github Releases? Thanks! I'm more than capable of building a static version myself but the time and effort to do so is a bit large for me right now.

aykevl, 2020-04-20:
There are already (mostly) static builds in GitHub Releases. Do they not work on your system?

prologic, 2020-04-20:
> There are already (mostly) static builds in GitHub Releases. Do they not work on your system?

Yeah the last release I tried didn't work; I don't think they're completely statically linked? I can try the latest release and report back...

aykevl, 2020-04-26:
No they aren't statically linked, but the most important library we use (LLVM) is statically linked.
What system are you running that all these libraries are not available? Especially this error confuses me:

```
Error loading shared library ld-linux-x86-64.so.2: No such file or directory (needed by ./bin/tinygo)
```

prologic, 2020-04-27:
> What system are you running that all these libraries are not available? 

[uLinux](https://ulinux.org)

aykevl, 2020-04-29:
Reading that page (from Google Cache because the website is down) it doesn't seem like a distribution for normal usage - it doesn't even seem to support modern compilers (GCC/Clang). I'm not sure we can support such exotic systems.

That said, I do think it is worthwhile to try to get the TinyGo release builds linked entirely statically: it would avoid a number of issues with making the binary portable across Linux distributions (see #342 and #1064 for example).
I'm investigating that option by building TinyGo on Alpine Linux (which AFAIK doesn't link statically by default but makes that a whole lot easier).

prologic, 2020-04-29:
> Reading that page (from Google Cache because the website is down) it doesn't seem like a distribution for normal usage - it doesn't even seem to support modern compilers (GCC/Clang). I'm not sure we can support such exotic systems.

Sorry about the "offline"-ness. I'm rebuilding the VM that runs the Website.

And yes you are right. It is not designed for "everyday" use per se -- certainly not for Desktop use. I am the author and I am primarily designing it to directly replace RancherOS as the primary container distirbution. Which it already does quite well.
 
> That said, I do think it is worthwhile to try to get the TinyGo release builds linked entirely statically: it would avoid a number of issues with making the binary portable across Linux [...]

This would be great. and the reason I raise this issue. Thank you for considing this!

aykevl, 2020-05-01:
Unfortunately, it appears impossible to use libclang with a statically linked musl. I get the following error:

    Call to dladdr() failed

At [these lines](https://github.com/llvm/llvm-project/blob/060f2f815c4d337f6a1ed9d582a8583b7ed9135e/clang/tools/libclang/CIndexer.cpp#L129-L130):

```c
  if (dladdr((void *)(uintptr_t)clang_createTranslationUnit, &info) == 0)
    llvm_unreachable("Call to dladdr() failed");
```

It is called from [here](https://github.com/llvm/llvm-project/blob/060f2f815c4d337f6a1ed9d582a8583b7ed9135e/clang/tools/libclang/CIndex.cpp#L3559).

Maybe this is fixable in libclang, I'm not sure.

rockwotj, 2023-06-14:
It looks like they've added a fallback in newer versions of clang for statically linked builds in musl: https://github.com/llvm/llvm-project/blob/10e7b451ada5ff87adb2e04cc6a21b5305e90b12/clang/tools/libclang/CIndexer.cpp#L136-L137

rockwotj, 2023-06-14:
Looks like it's supported in LLVM 15, @aykevl added it :)

https://github.com/llvm/llvm-project/commit/75d12e49c729e65f77bfdfe65b16df8c39944d35

aykevl, 2023-06-16:
TinyGo linux/amd64 binaries have been statically linked for a while now, see: https://github.com/tinygo-org/tinygo/pull/2826
Unfortunately, other architectures (like arm64) aren't statically linked yet.

prologic, 2023-06-16:
> TinyGo linux/amd64 binaries have been statically linked for a while now, see: https://github.com/tinygo-org/tinygo/pull/2826
Unfortunately, other architectures (like arm64) aren't statically linked yet.

This is great to hear 👌