tinygo-org/tinygo #1064: Document libstdc++ version dependency GLIBCXX_3.4.22

candlerb, 2020-04-20:
At https://tinygo.org/getting-started/linux/#ubuntu-debian it says:

> If you are using Ubuntu or another Debian based Linux on an Intel processor, download the DEB file from Github and install it using the following commands

I tried it under Ubuntu 16.04 LTS, but got the following error:

```
root@builder:~/tiny# tinygo build hello.go
tinygo: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.22' not found (required by tinygo)
```

Checking the libstdc++ version:

```
root@builder:~/tiny# dpkg-query -l |grep stdc
ii  libstdc++-5-dev:amd64            5.4.0-6ubuntu1~16.04.12                    amd64        GNU Standard C++ Library v3 (development files)
ii  libstdc++6:amd64                 5.4.0-6ubuntu1~16.04.12                    amd64        GNU Standard C++ Library v3

root@builder:~/tiny# ldd /usr/local/tinygo/bin/tinygo
/usr/local/tinygo/bin/tinygo: /usr/lib/x86_64-linux-gnu/libstdc++.so.6:
        version `GLIBCXX_3.4.22' not found (required by /usr/local/tinygo/bin/tinygo)
	...

root@builder:~/tiny# strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_DEBUG_MESSAGE_LENGTH
root@builder:~/tiny#
```

I tested Ubuntu 18.04 and it works.

Conclusion: the tinygo deb package requires a relatively recent version of Ubuntu/Debian.  It would be helpful if this could be mentioned, e.g. at least

> The package requires a system with `libstdc++6` providing version `GLIBCXX_3.4.22`

Then at least if you see this error, you know you're running on an unsupported version of the OS.  (Maybe there's a better way of checking the version than the "strings" command I showed?)

aykevl, 2020-04-20:
Well Ubuntu 16.04 is kind of old. Unfortunately we miss it by just one glibcxx version.

I'm thinking it might be possible to somehow do a completely static build using musl, but I'm not sure how painful that is to configure or whether it works at all. We already do the builds on an old Debian version (stretch) to be compatible with most distros but unfortunately it's hard to be portable to everything.

candlerb, 2020-04-20:
Maybe just add a note saying that GLIBCXX_3.4.22 is required, and Ubuntu 16.04 is too old.  (16.04 is still supported for another year).

BTW, I only have one Debian machine to test. That is running 9.12, and "strings" shows that versions up to and including `GLIBCXX_3.4.22` are present, so it should be fine.  Debian 8 is technically still in LTS but that ends in a couple of months, on 30 June 2020.

deadprogram, 2020-04-20:
That is weird, because I tested on exactly that version Ubuntu 16.04, and did not have that issue. I wonder what I installed along the way to make it work?

candlerb, 2020-04-20:
[Here](https://askubuntu.com/questions/575505/glibcxx-3-4-20-not-found-how-to-fix-this-error) it was suggested to add [ppa:ubuntu-toolchain-r/test](https://launchpad.net/~ubuntu-toolchain-r/+archive/ubuntu/test) - had you done that by any chance?

FWIW, I don't need 16.04 support - it was easy enough to switch to an 18.04 box, once I had worked out what the problem was.  I just thought it would be nice to help others avoid the same mistake.

closed 2024-11-30 by deadprogram

aykevl, 2024-12-01:
Yes, at least for Linux/amd64 we have statically linked binaries so this problem shouldn't happen anymore.