tinygo-org/tinygo #1065: Go module support

cleahcim, 2020-04-20:
Go modules is the recommended way to develop now since go1.14 ( recently released ).  It seems that tinygo only works the "old way" ( but still valid ) by developing in your GOPATH.

What is needed to make tinygo work with go.mod?

deadprogram, 2020-04-20:
Hi @cleahcim please take a look at https://github.com/tinygo-org/tinygo/pull/941

aykevl, 2020-05-13:
I believe #941 is now ready for review. You could try that branch and see whether it works for you.

ski7777, 2020-08-06:
Can someone explain #941 to me? How can I use tinygo as go module?
Thanks

aykevl, 2020-08-06:
@ski7777 it doesn't work that way. You don't use TinyGo as a Go module. Instead, #941 enables TinyGo to support Go modules, that is, it will read go.mod and go.sum when it compiles firmware for a microcontroller for example.

aykevl, 2020-08-06:
Closing as this has been fixed in the latest release.

closed 2020-08-06 by aykevl