tinygo-org/tinygo #1066: Build errors in 0.13 but not 0.12

codyoss, 2020-04-21:
## Build info

Target board: arduino nano
Version of tinygo: 0.13

##Background:

I was trying to compile an example from here: https://github.com/tinygo-org/drivers/blob/master/examples/ssd1306/i2c_128x32/main.go
I was using a modified version of the `ssd1306` to remove SPI methods as their is not support for my nano.

## What I did

## 0.12

### Command 
`docker run --rm -v $(pwd):/src tinygo/tinygo:0.12.0 tinygo build -o /src/oled.hex -size=short -target=arduino-nano /src/main.go`

### Output 

```
   code    data     bss |   flash     ram
   4384     192     658 |    4576     850
```

## 0.13

### Command

`docker run --rm -v $(pwd):/src tinygo/tinygo:0.13.0 tinygo build -o /src/oled.hex -size=short -target=arduino-nano /src/main.go`

### Output

```
/tmp/tinygo218597673/extra-0-scheduler_avr.S.o: In function `tinygo_startTask':
(.text.tinygo_startTask+0x6): undefined reference to `tinygo_pause'
collect2: error: ld returned 1 exit status
error: failed to link /tmp/tinygo218597673/main: exit status 1
```

jaddr2line, 2020-04-21:
This is a known issue and will be fixed in 0.13.1 which will be released soon.

codyoss, 2020-04-21:
Is this a duplicate of another issue? I can close if so.

jaddr2line, 2020-04-21:
No, no issue was made for it, and it went straight to pull requests in #1058 and #1060.

jaddr2line, 2020-04-21:
Version 0.13.1 is currently being released, so I am going to close this.

closed 2020-04-21 by jaddr2line