tinygo-org/tinygo #1075: MIPS architecture support

GTANAdam, 2020-04-27:
It seems that tinygo doesn't not support MIPS based embedded devices out of the box, although LLVM does support it.

jaddr2line, 2020-04-27:
As far as I can tell, there are only a few things required to add support for such a target:
1. a runtime arch file - provides the equivalent GOARCH value, bitness of the cpu, etc. (example [here](https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/src/runtime/arch_386.go))
2. a mapping from the LLVM arch name to the GOARCH name [here](https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/compileopts/target.go#L198)
3. helper program selection guesses [here](https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/compileopts/target.go#L282)

I can't think of anywhere else where changes would be necessary. After the appropriate changes are added, you should be able to build for MIPS by supplying an appropriate LLVM target to the `-target` flag.

deadprogram, 2020-04-27:
Is there specific MIPS hardware you are looking to support?

jaddr2line, 2020-04-27:
@deadprogram This issue seems to be regarding a Linux target, rather than baremetal.

GTANAdam, 2020-04-27:
> Is there specific MIPS hardware you are looking to support?

Yeah, It would be great to have bare metal support for mips24kc out of the box since that's basically used by most routers.
I'll try tinkering around, might learn something from it.

aykevl, 2020-04-27:
> Yeah, It would be great to have bare metal support for mips24kc out of the box since that's basically used by most routers.

> My intent is to execute the tinygo compiled binary within a mipsel linux box, no bare metal support, so I assume that's feasible considering llvm support.

Are you interested in baremetal or Linux support?
The two are quite different. Running under Linux would be relatively easy, needing basically the changes that @jaddr2line mentioned above. Baremetal support will take a lot more work and will usually be very chip specific. To get an idea of what is needed, see [this PR](https://github.com/tinygo-org/tinygo/pull/392/files) (but note that quite a few things changed since then to avoid duplicated code).

> It seems that after manually building LLVM as mentioned in [Getting Started](https://tinygo.org/getting-started/linux/#with-a-self-built-llvm) doesn't contain mips arch as a target

Yes, you need to modify this line to add Mips to the enabled targets:
https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/Makefile#L157

GTANAdam, 2020-04-28:
Thank you for the tip, I was already digging inside, getting that little mipsy mips support in LLVM 😄 

> Are you interested in baremetal or Linux support?
I am honestly interested in having it work in both Linux and bare metal but I do understand that bringing bare metal support is a bit out of my skill scope, but I am very eager to learn and contribute

olarriga, 2020-04-28:
I'm also interrested on compiling for MTK7628NN with Linux support. With Go I select GOARCH=mipsle GOMIPS=softfloat.
It's possible with TinyGo ? Do you have any documents ?
Thanks

jaddr2line, 2020-04-28:
@olarriga this is not yet implemented, see my first comment on this issue

GTANAdam, 2020-04-28:
> I'm also interrested on compiling for MTK7628NN with Linux support. With Go I select GOARCH=mipsle GOMIPS=softfloat.
> It's possible with TinyGo ? Do you have any documents ?
> Thanks

Yes, it's possible and beautiful, I'm currently working on it, will have a PR and a documentation published soon.

aykevl, 2020-04-29:
> Currently the compilation is spewing a warning concerning wrong floating point setting but the binary is functioning properly, yet to figure out where to put the -msoft-float flag in tinygo/llvm given that the musl toolchain already uses -msoft-float
> 
> ```
> /home/adam/toolchain/toolchain-mips_24kc_gcc-8.4.0_musl/bin/../lib/gcc/mips-openwrt-linux-musl/8.4.0/../../../../mips-openwrt-linux-musl/bin/ld: warning: /tmp/tinygo498244004/main uses -msoft-float (set by /home/adam/toolchain/toolchain-mips_24kc_gcc-8.4.0_musl/bin/../lib/gcc/mips-openwrt-linux-musl/8.4.0/../../../../mips-openwrt-linux-musl/lib/crt1.o), /tmp/tinygo498244004/main.o uses -mhard-float
> ```

You probably need to set a flag (`-soft-float` maybe?) in the `BuildTags` property. This list of flags is for how TinyGo compiles the binary, and is independent of `CFlags` (which is used to build the C parts). You can figure out the needed flag with the Clang `-v` flag (look for `-target-fetature`).
While some programs may work, I do not think programs that have float operations will work without this change. Try running testdata/float.go for example.

> After hours of thorough studying of mips specs and docs, I have found out that the /o32 ABI only has 4 argument registers which is less than syscall6 arg length thinking... additionally, the syscall manpage mentions that the syscall convention passes arguments 5+ through 8 on the user stack but that requires a different implementation in gollvm(?)..
> I am currently limited by my lack of knowledge in this area so any input would be appreciated! smile

Ah really, they are passing syscall arguments on the stack? That complicates things a bit.
I think the easiest option is to write assembly functions that do the syscall itself. Calling those functions will make sure LLVM puts those values on the stack in the right place. There is a bit of a complication however with the success/error value (in `$a3`) so the function may need to do a bit more than just call `syscall` and return.

luis33338, 2021-01-25:
Duplicate of #

ski7777, 2021-03-19:
Any update on this?

GTANAdam, 2021-03-20:
> Any update on this?

Unfortunately not, I have decided that writing code for MIPS based embedded devices is much better done via C/C++, I haven't had the time to figure it out, if one could provide some examples then I could probably try it out again.

dosgo, 2021-09-29:
Is there no reference value for the system call of the golang source code Go\src\syscall.  Although C/C++ is good, cross-compilation is troublesome, and development efficiency is not good. Rust solves the cross-compilation problem well.

mutexstream, 2022-05-31:
Any update on this? This will really be very beneficial for running go code on routers.

aykevl, 2022-05-31:
We could do it, but I haven't seen a lot of demand (except for this issue). I'm somewhat worried about the maintenance burden this will cause, because it would add yet another architecture. We currently support arm, arm64, 386, amd64, avr, xtensa, and wasm. But if there is enough demand or a good use case, this would certainly be interesting to support.

mutexstream, 2022-05-31:
Writing networking and server code for routers with go should be a very interesting use case. The main reason why go isn't even considered is because of the huge binary size, but if tiny go can be made to work with these routers, it will be a very good alternative to C/C++ for many cases.

Example:
https://mender.io/blog/why-did-we-choose-golang-over-c 
https://stackoverflow.blog/2022/04/04/comparing-go-vs-c-in-embedded-applications/

dosgo, 2022-09-21:
> We could do it, but I haven't seen a lot of demand (except for this issue). I'm somewhat worried about the maintenance burden this will cause, because it would add yet another architecture. We [...]

I think mips is much more important than 386 and amd64. Mips are only embedded devices such as routing. On the contrary, 386 amd64 devices have huge space and can use official compilers.

vshymanskyy, 2022-09-21:
You could target wasm and then run wasm on mips...

aykevl, 2022-09-29:
So if we would add support for MIPS, which would it be? Apparently there are 4 MIPS architecture variants that we could support:

>     mips       - MIPS (32-bit big endian)
>     mips64     - MIPS (64-bit big endian)
>     mips64el   - MIPS (64-bit little endian)
>     mipsel     - MIPS (32-bit little endian)

I don't think 64-bit MIPS needs TinyGo, so that leaves it with `mips` and `mipsel`. I don't know which of the two architectures (big or little endian) is used more widely, but I would expect a little-endian system to be much easier to add support for since all architectures currently supported by TinyGo are little endian.

dosgo, 2022-09-29:
The mips endian seems to be OS dependent! https://openwrt.org/docs/platforms/start is little endian more.

mutexstream, 2023-02-06:
Was any decision made regarding this issue?

lePereT, 2023-02-17:
We (Jangala - a charity that makes internet access systems for schools, clinics and disaster zones around the world) would love to see this added.

We currently use Lua for our codebase but are thinking of using TinyGo on our MCUs and would love to use it on our Arm64 and MIPS based boards that are running OpenWrt/Linux. The Ath79 target (a hugely popular router and AP chipset family) is big endian.

If this became available it could be hugely popular amongst the OpenWrt community (who currently don't see Go as a viable option). For example, the Mender IoT remote package that @mutexstream [references](https://github.com/tinygo-org/tinygo/issues/1075#issuecomment-1142332317) generates a 2MB+ binary which is prohibitive on this kind of target.

vshymanskyy, 2023-02-17:
Wasm3 could be your saviour as well 😎

lePereT, 2023-02-18:
Wasm seems interesting but by necessity the code we write interacts with system libraries and the kernel directly, which I guess (not knowing a huge amount about it) might be more difficult with Wasm

aykevl, 2023-02-18:
I can look into adding support for MIPS. But again: which variant would be preferred? There are four to pick from for initial support. (Note: big endian is probably going to be more difficult than little endian).

lePereT, 2023-02-19:
Just looking at the number of end user devices that would be hit, I'd probably suggest the following order of priority:
  1. mipseb (32 bit big endian)
  2. mipsel (32 bit little endian)

The rationale for this is that just the [ath79 family](https://downloads.openwrt.org/snapshots/targets/ath79/generic/) alone represents probably hundreds of millions of targets devices worldwide

aykevl, 2023-02-19:
I did some experimenting with MIPS support and managed to get very limited support working:

```
$ GOARCH=mips tinygo run -gc=leaking ./testdata/alias.go
x
apple
true
false
```

I have no plans currently to continue working on this, but if there's anybody interested feel free to complete support for MIPS. The branch is here: https://github.com/tinygo-org/tinygo/compare/dev...mips

mutexstream, 2023-02-22:
What's left to do for full support? And do you have any docs that can help me learn how to add a new architecture?

aykevl, 2023-02-22:
> What's left to do for full support?

I was stuck at syscall support. There are probably a few more things but that's a big one.
To implement this, you need to understand how system calls work on MIPS (there are two calling conventions and I don't know which one is usually used) and implement it using inline assembly in compiler/syscall.go. There are a number of other architectures you can look at for help.

> And do you have any docs or example pull requests that can help me learn how to add a new architecture?

Not really, except for looking at what's already there and the various MIPS documents. Examples: [syscall (2)](https://man7.org/linux/man-pages/man2/syscall.2.html), [wiki](https://www.linux-mips.org/wiki/Syscall), [instruction set](https://www.cs.cmu.edu/afs/cs/academic/class/15740-f97/public/doc/mips-isa.pdf), etc.

adonespitogo, 2023-03-19:
I also need this to run go on openwrt router. I agree that Mips is more important than amd64 and i386. The only deterrent for not using go for embedded is the binary size. Performance-wise, go is very good alternative to C/C++.

b0ch3nski, 2023-09-20:
Since `netdev` in #3704 is just around the corner, having a support for MIPS would open up a lot of possibilities for usage in routers (looking at you, OpenWRT guys :slightly_smiling_face:).

adonespitogo, 2023-09-21:
This is very timely. Our project is still in development and we're using the standard go compiler shipped with openwrt.

nonibytes, 2023-10-28:
When will this be done? I am very interested in this feature!

gtxaspec, 2024-02-28:
would love to see mips support!

wltechblog, 2024-06-24:
Just to chime in here, I'm part of a project that builds open source replacement firmware for MIPS (Ingenic T series, 32 bit mips, little endian)  based IP cameras which are quite popular. These devices tend to have 8MB flash to contain the bootloader, kernel, userland, and configs.. So regular Go is out of the question for them as a simple "Hello, World" executable in Go is 3x larger than our video capture and streaming server. Being able to write in Go would open a lot of possibilities for us but for the moment we're all grumpily going back in time a few decades and using C. 

To my knowledge, Ingenic is the only chipmaker actively producing and developing MIPS cores today.

aykevl, 2024-06-24:
I need a test environment to be able to do this. Right now, QEMU has bugs which prevent me from doing this: https://gitlab.com/qemu-project/qemu/-/issues/2082

wltechblog, 2024-06-24:
> I need a test environment to be able to do this. Right now, QEMU has bugs which prevent me from doing this: https://gitlab.com/qemu-project/qemu/-/issues/2082
> 
> EDIT: I found a workaround, I might be able to do this.

I can provide remote access to a mips system if it would be helpful.

aykevl, 2024-06-24:
I found a workaround to be able to use QEMU.
Remote access is not really useful since it slows down testing and makes automatic testing super painful.

---

In other news, I have a "hello world" MIPS program working: https://github.com/tinygo-org/tinygo/pull/4308

aykevl, 2024-06-25:
#4308 should work in a lot of cases. For anybody with actual hardware: can you test how well this PR works? (Qemu might behave differently than actual hardware for example).

wltechblog, 2024-06-25:
I was able to build a small "Hello, World!" app and run it on one of my MIPS based ip cameras:

```[root@ing-wuuk-y0510-4238 tmp]# uname -a && ./hello
Linux ing-wuuk-y0510-4238 3.10.14__isvp_swan_1.0__ #1 PREEMPT Mon Jun 24 00:01:45 UTC 2024 mips GNU/Linux
Welcome to TinyGo!
```
I will do some more testing (need to try to use musl dynamic instead of static linking if possible).

aykevl, 2024-06-25:
Great!

> need to try to use musl dynamic instead of static linking if possible

That's not really supported right now. It may be possible to support in the future but will take some work.

Can you explain why you need dynamic linking?
If it's to reduce binary size: in my experience it has very little impact or will actually _increase_ binary size (because the dynamic linker needs to be included).

wltechblog, 2024-06-25:
Indeed the goal is always to reduce file size on these little machines.

As it stands, a normal hello world (using fmt) with debugging disabled is 234k and can be compressed on storage with upx down to 68k. Skipping fmt and using the println function, 13k after upx compression. 

This is far far better than the native Go compiler (about 850kb after upx compression using the println function)!

I understand dynamic linking may not help after all, and clearly being very strict with package imports (and their dependencies) will be necessary. I'll continue testing with additional programs, i'm curious to see how the memory footprint looks as our typical target platform has 8mb flash and 40-ish mb ram after reserving a chunk for hardware devices!

aykevl, 2024-06-25:
There are various ways to reduce binary size further, please take a look here:
https://tinygo.org/docs/guides/optimizing-binaries/
Most importantly, you'll want to strip debug information from the binary to reduce the size.

wltechblog, 2024-06-25:
Ah, I expected the -no-debug flag to do the same job as strip, but I see that is not the case.  Executable sizes before upx compression dropped by around half when stripping an exe that was build with -no-debug.

lePereT, 2024-06-26:
> #4308 should work in a lot of cases. For anybody with actual hardware: can you test how well this PR works? (Qemu might behave differently than actual hardware for example). Here you can find instructions how to build TinyGo: https://tinygo.org/docs/guides/build/

Would love to test but only have Big Endian hardware (QCA9531 series and will be happy to test comprehensively for that target)

aykevl, 2024-06-26:
@lePereT I've added big endian support. A few things (like maps, reflect, cgo) don't fully work yet, probably because it's the first big-endian system in TinyGo and there are existing bugs. But it would be helpful to know how well it already works.

aykevl, 2024-06-26:
@wltechblog I did a quick test and the statically linked C library is ~5kb for a very simple program. For a larger program it can go up to maybe 17kb or so.

aykevl, 2024-07-21:
https://github.com/tinygo-org/tinygo/pull/4308is now ready. For those interested in MIPS support: can you test how well this works for you? (Note that GOOS=mips isn't fully supported yet though it may work for you, only GOOS=mipsle is actually tested in CI and is supposed to work apart from 64-bit atomics).

DJAGABATWORK, 2024-07-29:
I used the dev branch version in a docker environment to build the serial example for a Teltonika TRB-255. This device has a QCA9531 with big endian support and emulated floating point support. When I issue the following command, an error occurs on The Teltonika, running the example just compiled with: 

docker run --rm -e GOARCH=mips -e GOOS=linux -e GOMIPS=softfloat -v ${pwd}:/src tinygo/tinygo-dev:latest tinygo build -o /src/serial examples/serial

The generated binary has the following structure:
serial: ELF 32-bit MSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), statically linked, with debug_info, not stripped

On the Teltonika the following message occurs when starting the serial example:
Illegal instruction

I should start with a simple hello world example. Can anybody show the right docker command to compile a custom project which is not "inside" the tinygo docker image but on the host system?
Thanks.

aykevl, 2024-07-29:
> On the Teltonika the following message occurs when starting the serial example:
> Illegal instruction

Can you run a debugger on the board to see which instruction it hits?

> Can anybody show the right docker command to compile a custom project which is not "inside" the tinygo docker image but on the host system?

I don't have much experience with Docker, I usually just run the binaries directly (or compile TinyGo myself). I think you can download them from here for example: https://github.com/tinygo-org/tinygo/actions/runs/10042659976 (I can, but I'm not sure whether non-team-members can download these binaries).

DJAGABATWORK, 2024-07-29:
@aykevl. Thanks for your reply. Unfortunately the only debugging means that are at my disposal are logread and dmesg for system and kernel messages. Neither of them show relevant information that could lead to the cause of the Illegal Instruction message.

I will have a look at the link, thank you.

Does TinyGo use the -e MIPS=softfloat option I used in de docker command or is it ignored?

aykevl, 2024-07-29:
> Does TinyGo use the -e MIPS=softfloat option I used in de docker command or is it ignored?

Not yet, I'm working on it.

b0ch3nski, 2024-07-29:
> My router runs on MediaTek MT7621AT SoC and "big Go" compiles perfectly fine for it using `GOARCH=mipsle` flag.

I can confirm that current dev branch of TinyGo successfully builds the binary that works on the above HW. Great job @aykevl !

The super basic test - create `main.go`:
```golang
package main

func main() {
	println("hello world")
}
```

Build with Docker:
```
$ docker run --rm --name "tinygo-build" -v "$(pwd):/src" -w "/src" -e GOARCH=mipsle -e GOOS=linux tinygo/tinygo-dev:latest tinygo build -no-debug -o test-mipsle main.go
...
$ file test-mipsle
test-mipsle: ELF 32-bit LSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), statically linked, not stripped
```

What's interesting is the file size difference - when I build the `amd64` binary exactly the same, `mipsle` is more than twice the size:
```
$ du -b test-*
15200	test-amd64
33408	test-mipsle
```

But after running `upx` on both, they get quite similar in size:
```
$ du -b test-*
12624	test-amd64
11388	test-mipsle
```

aykevl, 2024-07-29:
@DJAGABATWORK this might fix the issue: https://github.com/tinygo-org/tinygo/pull/4371

aykevl, 2024-07-29:
@b0ch3nski I don't know why that is, my guess is that MIPS is an even less efficient instruction set than amd64 (or it hasn't gotten a lot of optimization work).

To reduce binary size, you can also try `strip` (or `llvm-strip`) which strips unnecessary stuff like the symbol table. The binary size after stripping would be a more fair comparison.

DJAGABATWORK, 2024-07-30:
@b0ch3nski 's Docker example got me on the right track. Thank you.
I entered:
docker run --rm --name "tinygo-build" -v "$(pwd):/src" -w "/src" -e GOARCH=mips -e GOOS=linux -e GOMIPS=softfloat tinygo/tinygo-dev:latest tinygo build -no-debug -o test-mips main.go

Running from the same directory were the main.go file is.

contents of main.go

package main

func main() {

        println("Hello from here")
}

file test-mips
test-mips: ELF 32-bit MSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), statically linked, not stripped

Execution results:

root@Teltonika-TRB255:/installs# ./test-mips
Hello from here

So it is working with simple Hello World example on big endian mips. too! Good job @aykevl 
In the coming weeks I will increase complexity to see if I can get it to crash again, like the serial example does.
Btw @aykevl: I tested the serial binary again after your change mentioned in #4371. It unfortunately still crashes with Illegal Instruction message, but without further details this info is of little use. Slowly increasing complexity from the hello world  example is the way to (tiny)go IMO.

aykevl, 2024-07-30:
@DJAGABATWORK

> I tested the serial binary again after your change mentioned in #4371.

How did you test it? It's not merged, so if you used Docker you weren't using that change.
It would be really useful to know the faulting instruction. Does it print an address somewhere? If you give me the fault address and the binary I can probably figure out what went wrong.

(Also I formatted your comment a little, I hope you don't mind).

aykevl, 2024-07-30:
So I found that Go normally uses MIPS32r1 as the minimum requirement, while TinyGo uses MIPS32r2. That could be a reason why it crashes.
@DJAGABATWORK do you know which specific CPU core this is? All I can see on the Qualcomm website is "Clock Speed: Up to 650 MHz" which is useless information for this purpose.

aykevl, 2024-07-31:
> Looks like this exact SoC doesn't have FPU but binaries build with both big Go and latest TinyGo dev runs just fine without `GOMIPS=softfloat` 🤔

You probably just got lucky. A hello world example is unlikely to use FPU instructions, and appears to also have avoided the new instructions introduced in mips32r2.

After the softfloat PR is merged, I intend to make a new PR that switches to mips32r1 (aka mips32).

[18 more replies]

closed 2024-11-27 by deadprogram