tinygo-org/tinygo #1076: flash: getDefaultPort() fails on Windows locales such as Japan

sago35, 2020-04-27:
TinyGo on Windows uses wmic in the following, but wmic can't get it well because it returns different strings by locale.
https://github.com/tinygo-org/tinygo/blob/master/main.go#L606.

In English, it probably returns a string like `USB Serial Device (COM7)`, but in Japanese, it returns a string like `USB シリアル デバイス (COM7)`.
I think the solution is as follows.

1. wmic PATH Win32_SerialPort where "Caption LIKE 'USB Serial%' OR Caption LIKE 'USB シリアル%'" GET DeviceID
2. wmic PATH Win32_SerialPort where "Caption LIKE 'USB % (COM%)'" GET DeviceID
3. wmic PATH Win32_SerialPort where "Caption LIKE 'USB %'" GET DeviceID
4. reg query HKLM\HARDWARE\DEVICEMAP\SERIALCOMM

1 needs to add a string condition, such as German or French.
2 and 3 may be written differently in German or French.
4 is a string that works correctly in my environment (windows10 pro) and is probably not affected by locale, but I don't know the details.

    $ reg query HKLM\HARDWARE\DEVICEMAP\SERIALCOMM

    HKEY_LOCAL_MACHINE\HARDWARE\DEVICEMAP\SERIALCOMM
        \frzeroDevice\frzeroUSBSER000 REG_SZ COM7

I think 2 is better, but I'm not sure it works properly in non-Japanese locales (even in English).

sago35, 2020-04-27:
I've created a commit for 1 to 3.
I haven't been able to decide which one is better.
Do you have a good idea?

https://github.com/sago35/tinygo/tree/getDefaultPort1
https://github.com/sago35/tinygo/tree/getDefaultPort2
https://github.com/sago35/tinygo/tree/getDefaultPort3

If possible, I would like French, German, or other Windows users to try the following

    wmic PATH Win32_SerialPort

sago35, 2020-04-27:
memo:
windows - Communicating with non-English WMI - Stack Overflow
https://stackoverflow.com/questions/28989279/communicating-with-non-english-wmi

sago35, 2020-04-27:
english : `USB Serial Device (COM7)`
japanese : `USB シリアル デバイス (COM7)`

(picture here)
[image]

sago35, 2020-04-27:
In my environment, the subsequent touchSerialPortAt1200bps() returns an err even if wmic succeeds.

Changing BaudRate: 1200 to BaudRate: 9600 will eliminate the error, but it probably doesn't make sense in TinyGo.

windows 10 pro
go version go1.13.8 windows/amd64
feather-m4

Test code is below.
https://gist.github.com/sago35/44c11bf100edee171f5c52647b9b215b

aykevl, 2020-04-29:
We definitely need to fix this. I didn't expect these strings to change between different language variants of Windows.

The first variant is definitely a no-go. It would require adding support for every language that Windows uses, which is a nightmare to test and will probably still leave some people out (not to mention future Windows updates which might change these strings).
The second variant also seems a bit iffy, but might work. At least it doesn't hardcode a particular language. The third variant is probably too broad and might return devices that aren't serial devices.

For a reliable solution, I think we should either get the value directly from the source (like in point 4) or somehow convince `wmic` to return values in a specific language (such as English).

> Changing BaudRate: 1200 to BaudRate: 9600 will eliminate the error, but it probably doesn't make sense in TinyGo.

It has to be 1200 specifically, because that baud rate is interpreted in a special way: the USB-CDC implementation on the chip recognizes it as a special hint to enter the bootloader. Are you sure it was connecting to the Feather M4 and not some other connected hardware?

sago35, 2020-05-01:
> For a reliable solution, I think we should either get the value directly from the source (like in point 4) or somehow convince wmic to return values in a specific language (such as English).

I'll consider a better way.

> It has to be 1200 specifically, because that baud rate is interpreted in a special way: the USB-CDC implementation on the chip recognizes it as a special hint to enter the bootloader. Are you sure it was connecting to the Feather M4 and not some other connected hardware?

I created issue #1077.

sago35, 2020-05-01:
> For a reliable solution, I think we should either get the value directly from the source (like in point 4) or somehow convince wmic to return values in a specific language (such as English).

I tried to read from the registry with the following code.

environment:

| port | caption | comment |
| ---- | -------- | ----------- |
| COM3 | USB シリアル デバイス (COM3) | PyPortal |
| COM12 | USB シリアル デバイス (COM12) | feather m4 |
| COM18 | USB Serial Port (COM18) | FTDI FT232RL (not TinyGo device) |

The execution results are as follows.
The last connected or reset device will be the last to be displayed.
Also, 'USB Serial Port (COMxx)', which is not displayed in wmic path win32_serialport, will be displayed.

In most cases, I think the last port connected or reset is the port to be developed, but I'm not sure.

```
$ main.exe
COM18
COM12
COM3
```

code:

```go
package main

import (
	"fmt"
	"log"

	"golang.org/x/sys/windows/registry"
)

func main() {
	ports, err := getSerialPorts()
	if err != nil {
		log.Fatal(err)
	}

	for _, port := range ports {
		fmt.Println(port)
	}
}

func getSerialPorts() ([]string, error) {
	k, err := registry.OpenKey(registry.LOCAL_MACHINE, `HARDWARE\DEVICEMAP\SERIALCOMM`, registry.QUERY_VALUE)
	if err != nil {
		fmt.Println(1)
		return nil, err
	}
	defer k.Close()

	names, err := k.ReadValueNames(0)
	if err != nil {
		fmt.Println(2)
		return nil, err
	}

	ports := []string{}
	for _, name := range names {
		//fmt.Println(name)
		s, _, err := k.GetStringValue(name)
		if err != nil {
			return nil, err
		}
		ports = append(ports, s)
	}

	return ports, nil
}

// reg query HKLM\HARDWARE\DEVICEMAP\SERIALCOMM
```

aykevl, 2020-05-02:
I checked `HKLM\HARDWARE\DEVICEMAP\SERIALCOMM` and indeed I think it would be the best way to query serial ports.

> In most cases, I think the last port connected or reset is the port to be developed, but I'm not sure.

I think it's best to just show an error instead. You can pick a particular serial port using the `-port` flag.

sago35, 2020-05-06:
There was a go.bug.st/serial.GetPortsList(), so I think it's better to use that than the one I implemented.
The implementation looks pretty much the same.

https://github.com/bugst/go-serial/blob/640e755b9433574aa3e73b230d98c187c962e85d/serial_windows.go#L30

```go
package main

import (
	"fmt"
	"log"

	"go.bug.st/serial"
)

func main() {
	ports, err := serial.GetPortsList()
	if err != nil {
		log.Fatal(err)
	}

	if len(ports) == 0 {
		fmt.Println("No serial ports found!")
	}

	for _, port := range ports {
		fmt.Println(port)
	}
}
```

sago35, 2020-05-10:
@aykevl 
Since #1084 is now merged, I think it's okay to close it.

PR #1084 (2020-05-07): flash: fix getDefaultPort() fails on Windows locales such as Japan
Resolves #1076
closed 2020-05-10 by aykevl