tinygo-org/tinygo #1077: flash: touchSerialPortAt1200bps() always returns an error.

sago35, 2020-05-01:
In my environment, touchSerialPortAt1200bps() always returns an error.
However, touchSerialPortAt1200bps() is used to transition to boot.
After the transition to boot, touchSerialPortAt1200bps() does not return an error.

Test:
windows10 pro
feather-m4

sago35, 2020-05-01:
When I add a debug print to touchSerialPortAt1200bps(),

```go
func touchSerialPortAt1200bps(port string) (err error) {
	retryCount := 3
	for i := 0; i < retryCount; i++ {
		// Open port
		p, e := serial.Open(port, &serial.Mode{BaudRate: 1200})
		if e != nil {
			fmt.Println("DEBUG:", i, e.Error()) // <<-- debug print
			time.Sleep(1 * time.Second)
			err = e
			continue
		}
		defer p.Close()

		p.SetDTR(false)
		return nil
	}
	return fmt.Errorf("opening port: %s", err)
}
```

I get something like this.
When serial.Open() returns "Invalid serial port", we are transitioning to boot.

```
DEBUG: 0 Invalid serial port
DEBUG: 1 Serial port not found
DEBUG: 2 Serial port not found
opening port: Serial port not found
```

sago35, 2020-05-01:
Only the feather-m4 is connected to the USB.
The COM port is set to the feather-m4 port.

sago35, 2020-05-04:
In TinyGo, go.bug.st/serial.Open returns an error because it is resetting immediately at the following location.
It returns an error, but it transitions to boot.

https://github.com/tinygo-org/tinygo/blob/dev/main.go#L480

When using arduino software, serial.Open is not an error.
So tinygo flash succeeds.
The reason is that it is reset after 250 ticks.

https://github.com/adafruit/ArduinoCore-samd/blob/master/cores/arduino/USB/CDC.cpp#L125

```c++
bool CDC_Setup(USBSetup& setup)
{
    // ...
    if (requestType == REQUEST_HOSTTODEVICE_CLASS_INTERFACE)
    {
        // ...
        if (r == CDC_SET_LINE_CODING || r == CDC_SET_CONTROL_LINE_STATE)
        {
            // auto-reset into the bootloader is triggered when the port, already
            // open at 1200 bps, is closed. We check DTR state to determine if host 
            // port is open (bit 0 of lineState).
            if (_usbLineInfo.dwDTERate == 1200 && (_usbLineInfo.lineState & 0x01) == 0)
            {
                initiateReset(250);  // <<-- !!! HERE !!!
            }
            else
            {
                cancelReset();
            }
            return false;
        }
        // ...
    }
    return false;
}
```

sago35, 2020-05-04:
I think there are two ways to fix it.

1. wait 250 tick and then reset according to the way arduinoCore does it.
    * SysTick_Handler needs to be enabled and will consume a little more resources
2. Ignore the InvalidSerialPort immediately after opening at 1200 bps, as shown in the following code
    * If you are not transitioning to boot here, flashUF2UsingMSD() will fail.

```go
func touchSerialPortAt1200bps(port string) (err error) {
    retryCount := 3
    for i := 0; i < retryCount; i++ {
        // Open port
        p, e := serial.Open(port, &serial.Mode{BaudRate: 1200})
        if e != nil {

            // !!! PATCH start !!!
            se, ok := e.(*serial.PortError)
            if ok && se.Code() == serial.InvalidSerialPort {
                // InvalidSerialPort error occurs when transitioning to boot
                return nil
            }
            // !!! PATCH end !!!

            time.Sleep(1 * time.Second)
            err = e
            continue
        }
        defer p.Close()

        p.SetDTR(false)
        return nil
    }
    return fmt.Errorf("opening port: %s", err)
}
```

aykevl, 2020-05-07:
I can reproduce this. I think delaying the reset is the correct approach. I wouldn't use the SysTick for this but instead the RTC, which is already used for timekeeping in the runtime. The RTC has two 32-bit compare registers, of which we're only using one. The other one could be used for the delay itself. Take a look here for example:

https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/src/runtime/runtime_atsamd51.go#L272

aykevl, 2020-05-08:
> The RTC has two 32-bit compare registers, of which we're only using one

After looking into this, I found that this is only true for the atsamd51 chips, not for atsamd21. That means a different solution is needed for the atsamd21.

sago35, 2020-05-12:
Is this problem only for windows?
Or is it the same for macos and linux?

sago35, 2020-05-12:
In the future, if goroutine's scheduling changes, it may not work, but the problem was solved by doing the following

```go
func cdcSetup(setup usbSetup) bool {
    // ...
    if setup.bmRequestType == usb_REQUEST_HOSTTODEVICE_CLASS_INTERFACE {
        // ...
        if setup.bRequest == usb_CDC_SET_LINE_CODING || setup.bRequest == usb_CDC_SET_CONTROL_LINE_STATE {
            // auto-reset into the bootloader
            if usbLineInfo.dwDTERate == 1200 && usbLineInfo.lineState&usb_CDC_LINESTATE_DTR == 0 {
                go func() {
                    ResetProcessor()
                }()
            } else {
                // TODO: cancel any reset
            }
            sendZlp()
        }
        // ...
    }
    // ...
}
```

I wish I could write the following.
But now you can't use the time package in the machine package.

```go
go func() {
    time.Sleep(250 * time.Millisecond)
    ResetProcessor()
}()
```

aykevl, 2020-05-12:
I haven't seen this problem on Linux.

Starting a goroutine from an interrupt handler is not allowed because heap allocations are not allowed in an interrupt handler. It might have worked this time, but it will likely be disallowed entirely in the future. Therefore, this cannot be a solution. I think using a timer is the only solution here.

sago35, 2020-05-24:
So far, I don't know a good solution.

If possible, please include a provisional patch for windows in the next release.
As it is now, Windows can't do tinygo flash.
I think it's better to be able to do it than not to be able to do it.
https://github.com/tinygo-org/tinygo/issues/1077#issuecomment-623273635

@aykevl 
How about limiting the scope of the patch to runtime.GOOS?

I'm making a `tinygo-flash` command and using it.
but the official `tinygo flash` should be able to do the same thing.
https://github.com/sago35/tinygo-flash

sago35, 2020-05-28:
I've created a PR #1085 .

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram