tinygo-org/tinygo #1078: compiler: empty slice pointer is out of bounds

jaddr2line, 2020-05-02:
When slicing down to 0 cap (`s[cap(s):]`), TinyGo currently places an out of bounds pointer into the resulting slice header. If this slice header is stored in a tracked allocation for an extended period of time, this can cause an unrelated allocation to stay alive. Our GC is currently conservative, so this may be acceptable now, but this will need to be corrected before we can try implementing a precise collector.

Currently, `gc` passes through the original pointer when the result is an empty slice: https://play.golang.org/p/eDGa133ePTF

There is no valid way to use unsafe to access the backing array from an empty slice header, so we could alternatively generate a pointer to something else (e.g. the zero base). This would allow the backing array to be freed.

The first option is probably the cheapest to implement, and the compiler might be able to optimize it away effectively in most circumstances.

dgryski, 2024-10-22:
Concrete example of the above Go code printing different pointer addreses:
```
package main

import (
	"fmt"
)

func main() {
	x := make([]byte, 5)

	fmt.Printf("%p\n", x)
	fmt.Printf("%p\n", x[5:])
	fmt.Printf("%p\n", x[1:])
	fmt.Printf("%p\n", x[1:][1:])
	fmt.Printf("%p\n", x[1:][1:][1:])
	fmt.Printf("%p\n", x[1:][1:][1:][1:])
	fmt.Printf("%p\n", x[1:][1:][1:][1:][1:])
}
```
produces:
```
~/go/src/github.com/dgryski/bug/sliceptr $ go run main.go
0xc00009000a
0xc00009000a
0xc00009000b
0xc00009000c
0xc00009000d
0xc00009000e
0xc00009000e
~/go/src/github.com/dgryski/bug/sliceptr $ tinygo run main.go
0x1041514d0
0x1041514d5
0x1041514d1
0x1041514d2
0x1041514d3
0x1041514d4
0x1041514d5
```

Note the different pointer addresses.

aykevl, 2024-10-22:
We _could_ do this, but I don't think it's a big problem (few people setting the slice to be empty and not nil I think) and apart from keeping an unnecessary allocation alive I don't think it's a real problem for a precise GC either (at least not the way the precise GC is implemented in TinyGo).

Also, I'm not sure this is a valid optimization. See this for example: https://go.dev/play/p/j_R9qEZMob4
We can't set the pointer to nil (that would change the behavior of comparing the slice to nil), we have to use something else. Luckily the compiler does not allow comparing two slices directly, but it does allow access to the underlying pointer through [`Value.Pointer`](https://pkg.go.dev/reflect#Value.Pointer) in the reflect package. So if we use some other value like `unsafe.Pointer(uintptr(1))`, this would still be visible and might result in unexpected behavior.

davecheney, 2026-11-01:
I think this was fixed by 11c2b76e ("runtime (gc.blocks): move objHeader to the end").

The block collectors now store the object header after the object body. A pointer one past the end of an allocation therefore lands on that allocation's own header and can never reach the next object. Boehm was never affected because TinyGo builds it with `ALL_INTERIOR_POINTERS`.

I tried to write a failing test for this. The program below places a heap object directly after a slice's backing array, keeps only `buf[len(buf):]` alive, drops the object, and checks that its finalizer runs. It passes on current `dev` with `-target wasm` (precise GC), with or without the compiler change in #6082.

```go
package main

import (
	"runtime"
	"unsafe"
)

// A six word object plus its two word trailing header fills two four word
// blocks, so the next allocation starts right after the header.
const (
	word = unsafe.Sizeof(uintptr(0))
	size = 6 * word
)

type box struct{ x [size]byte }

var (
	keepSlice []byte
	keepArray []byte
	freed     int
	sink      int
	end       = int(size)
)

//go:noinline
func scrubStack(depth int) int {
	if depth <= 0 {
		return sink
	}
	var buf [64]int
	for i := range buf {
		buf[i] = depth + i
	}
	sink += buf[depth&63]
	return scrubStack(depth-1) + buf[0]
}

//go:noinline
func adjacent(p *box, end unsafe.Pointer) bool {
	return unsafe.Add(end, 2*word) == unsafe.Pointer(p)
}

//go:noinline
func allocAfterSlice() bool {
	buf := make([]byte, size)
	p := new(box)
	if !adjacent(p, unsafe.Add(unsafe.Pointer(&buf[0]), len(buf))) {
		return false
	}
	runtime.SetFinalizer(p, func(*box) { freed++ })
	keepSlice = buf[end:]
	return true
}

//go:noinline
func allocAfterArray() bool {
	arr := new([size]byte)
	p := new(box)
	if !adjacent(p, unsafe.Add(unsafe.Pointer(arr), len(arr))) {
		return false
	}
	runtime.SetFinalizer(p, func(*box) { freed++ })
	keepArray = arr[end:]
	return true
}

func place(alloc func() bool) {
	for i := 0; i < 100; i++ {
		if alloc() {
			return
		}
	}
	panic("could not place an object after the slice")
}

func main() {
	place(allocAfterSlice)
	place(allocAfterArray)
	for i := 0; i < 200 && freed < 2; i++ {
		sink += scrubStack(40)
		runtime.GC()
		runtime.Gosched()
	}
	if freed != 2 {
		println("freed", freed, "of 2")
		panic("an empty slice kept the next object alive")
	}
	println("ok")
}
```

I'm closing this as fixed and will close #6082 too. Please reopen if you know of a case this misses.

closed 2026-11-01 by davecheney