tinygo-org/tinygo #1082: How to add an Interrupt Handler on avr ?

glococo, 2020-05-06:
Hi, I'm learning Go and I am surprised of how much mcu it support. Congrats.

btw, I just wanted to implement my custom irq_USART0_RX but the only way I found was to:
1) comment "interrupt.New..." call in "func (uart UART) Configure(config UARTConfig)" function inside machine_atmega.go
2) write my own function, for example inside main:
interrupt.New( avr.IRQ_USART_RX, func( intr interrupt.Interrupt ){ 
    data:=avr.UDR0.Get()
    uart.WriteByte( byte(data) ) 
  })

BUT: With this procedure, I have to import "device/avr" and "runtime/interrupt", and modify unwanted stock libraries.

Is there a better option to add a custom interrupt handler ?
I was thinking of a new optional "Configure" option, like baudrate, but for the irq_func_handler.
Is that right ?

aykevl, 2020-05-07:
What are you trying to do? Redefining an interrupt is not possible: you can only have one handler for an interrupt.

glococo, 2020-05-08:
Thank Aykevl for reply me,
I want to use the AVR uart interrupt itself, without pooling the uart.Buffered() as used in echo example.

I was thinking of having INT_RX variable in UARTConfig struct, and when a call uart.Configure() is made, check if that variable is loaded with a func or not, and assign it properly.
Something like baudrate. 

What do you think ?

aykevl, 2020-05-08:
> I want to use the AVR uart interrupt itself, without pooling the uart.Buffered() as used in echo example.

I understand that, but what are you _really_ trying to do? Why do you need your own interrupt handler?

The goal in the TinyGo project is to make calls such as `.ReadByte()` block, as is common in Go. That way you do not need `.Buffered()` or `time.Sleep` to wait for the next byte. This will probably make it unnecessary to have a custom interrupt handler.

glococo, 2020-05-08:
Let's suppose the AVR is sleeping and is powered by a li-po.
I need a function call for different kind of wake up. Like in C.

I was also trying to define/append a type UART function in "my_main.go" like this:
func (uart machine.UART) Handle_Int( intr interrupt.Interrupt ) {...

and in machine_atmega.go this:
func (uart UART) Configure(config UARTConfig) {
  if config.BaudRate==0 { config.BaudRate = 9600 }

  if uart.Handle_Int != nil {
    interrupt.New( irq_USART0_RX, uart.Handle_Int )
  } else {
    // Register the UART interrupt.
    interrupt.New( irq_USART0_RX, func(intr interrupt.Interrupt) {
      // Read register to clear it.
...
...

but compiler output this:
invalid receiver machine.UART (type not defined in this package)

And it seems reasonable to not been able to add type functions out of scope.But I need to setup interrupt calls from my program or module. And not modifing tinyGo source files, will make me brake upgrades.

BTW: In this case, I could workaround it, On wakeup, avr will continue the infinite loop and I could check again buffered(), but I want to learn and understand TinyGo and look no back anymore to each platform C SDKs.

aykevl, 2020-05-08:
So if I understand you correctly, you want to put the chip in a low power state but still react quickly to incoming bytes on the serial port?
This is currently not possible, but will be possible once blocking I/O reads are implemented. When that is implemented, you will not need to write a custom interrupt handler because the MCU will go into low power state while it is waiting for the byte to arrive. So I'm afraid you either have to wait for this to be implemented or use the method used in the echo example.

> ```go
> func (uart machine.UART) Handle_Int( intr interrupt.Interrupt ) {
> ```

This is indeed not allowed in Go: methods must be defined in the same package as the named type.

glococo, 2020-05-09:
Hi Aykevl, The low power state is an example. 
What I want is, attach a function to the fired interrupt.
In C, you attach your function to the interrupt vector.
In Go, tinyGo already attached a defined function inside machine_atmega.go, which leave me the only option to keep pooling a buffer.

glococo, 2020-05-09:
I saw you and @deadprogram wrote the interrupt API in
https://github.com/tinygo-org/tinygo/issues/515

But I think, for AVR I could not attach mine function. :(

aykevl, 2020-05-12:
TinyGo aims for a higher level abstraction layer than C. Therefore you should usually not use interrupts directly but rely on the machine package for those features. The interrupt API is mainly intended for the machine package.
That is why I'm asking why you need the interrupt handler.

glococo, 2020-05-12:
Hi Aykevl,
I use to write interrupt driven software and leave Sleep() inside the for {} loop.
It is a good practice to use interrupt driven software, and to do not waste oops polling a buffer to check if a PIN has changed.
Same for UART, when a byte arrive, let fire "x" function.

Like JS callback. Let the user attach a callback function after a Promise got returned.

deadprogram, 2024-11-30:
Closing since AVR does have interrupt handler these days.

closed 2024-11-30 by deadprogram