tinygo-org/tinygo #1086: compile to wasm: unsupported instruction

neophob, 2020-05-07:
Hi, thanks for tinygo! I struggle using it in my env:

```
go version go1.14.2 darwin/amd64
tinygo version 0.13.1 darwin/amd64 (using go version go1.14.2 and LLVM version 10.0.1)
```

I try to compile the wasm version of my project (https://github.com/neophob/ganother-world) - while go works tinygo fails with:

```
tinygo build -o "./dist/lib.wasm" -target wasm ./wasm
# github.com/neophob/ganother-world/wasm
../../../../../../../usr/local/Cellar/tinygo/0.13.1/src/sync/pool.go:14:14: interp: unsupported instruction

traceback:
../../../../../../../usr/local/go/src/fmt/print.go:137:17:
  %0 = call %runtime._interface @"(*sync.Pool).Get"(%sync.Pool* @fmt.ppFree, i8* undef, i8* undef), !dbg !1753
../../../../../../../usr/local/go/src/fmt/print.go:218:17:
  %5 = call %fmt.pp* @fmt.newPrinter(i8* undef, i8* undef), !dbg !1759
logger/logger.go:57:36:
  %10 = call %runtime._string @fmt.Sprintf(i8* %5, i32 %6, %runtime._interface* %7, i32 %8, i32 %9, i8* undef, i8* undef), !dbg !1710
wasm/main.go:10:13:
  call void @"github.com/neophob/ganother-world/logger.Info"(i8* getelementptr inbounds ([33 x i8], [33 x i8]* @"github.com/neophob/ganother-world/wasm.init#1$string", i32 0, i32 0), i32 33, %runtime._interface* null, i32 0, i32 0, i8* undef, i8* undef), !dbg !1702
github.com/neophob/ganother-world/wasm/<init>:
  call void @"github.com/neophob/ganother-world/wasm.init#1"(i8* undef, i8* undef), !dbg !1700
make: *** [tiny] Error 1
```
I'm clueless ;)

aykevl, 2020-05-07:
Yes, you probably should just remove the logging code in the initialization code (what looks like a `init` function at wasm/main.go line 10). One of the most important optimizations of TinyGo is trying to resolve all initialization code at compile time, and the fmt package is far too complex for it.

neophob, 2020-05-07:
Thanks @aykevl it looks related to https://github.com/tinygo-org/tinygo/issues/627.

however - i just commented out the effective code (`logger.go:57`) - now compilation takes for ever (stopped compilation after 30 mins), I think this should fail, no?

I'll try to remove fmt to resolve this and keep this issue updated

aykevl, 2020-05-07:
Yeah that sounds like a bug, it will usually take just a few seconds and definitely not longer than one minute under normal circumstances.

neophob, 2020-05-07:
can i provide some help to get to root cause? strace dump?

aykevl, 2020-05-07:
What I usually do is to send a `SIGABRT` signal to the process. This will cause an abort and will make sure the Go runtime prints a stacktrace. Then when you know where it is stuck, you can add debug prints. At least, that's how I would debug this.

neophob, 2020-05-08:
here you go
```
bash-3.2$ tinygo build -o "./dist/lib.wasm" -target wasm ./wasm
SIGABRT: abort
PC=0x7fff64290ce6 m=0 sigcode=0

goroutine 0 [idle]:
runtime.pthread_cond_wait(0x8dc5f08, 0x8dc5ec8, 0x30000000000)
	/usr/local/go/src/runtime/sys_darwin.go:390 +0x39
runtime.semasleep(0xffffffffffffffff, 0xc000000000)
	/usr/local/go/src/runtime/os_darwin.go:63 +0x85
runtime.notesleep(0x8dc5cc8)
	/usr/local/go/src/runtime/lock_sema.go:173 +0xe0
runtime.stopm()
	/usr/local/go/src/runtime/proc.go:1828 +0xc0
runtime.findrunnable(0xc000046800, 0x0)
	/usr/local/go/src/runtime/proc.go:2360 +0xa0d
runtime.schedule()
	/usr/local/go/src/runtime/proc.go:2520 +0x2fc
runtime.park_m(0xc000192780)
	/usr/local/go/src/runtime/proc.go:2690 +0x9d
runtime.mcall(0x40614f6)
	/usr/local/go/src/runtime/asm_amd64.s:318 +0x5b

goroutine 1 [syscall]:
tinygo.org/x/go-llvm._Cfunc_LLVMRunFunctionPassManager(0x1573f910, 0x11a133d8, 0x0)
	_cgo_gotypes.go:8811 +0x4d
tinygo.org/x/go-llvm.PassManager.RunFunc.func1(0x1573f910, 0x11a133d8, 0x11a133d8)
	/Users/distiller/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200401165421-8d120882fc7a/ir.go:2004 +0x79
tinygo.org/x/go-llvm.PassManager.RunFunc(0x1573f910, 0x11a133d8, 0x1)
	/Users/distiller/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20200401165421-8d120882fc7a/ir.go:2004 +0x35
github.com/tinygo-org/tinygo/transform.Optimize(0x9f11980, 0xc0002d0060, 0x2, 0x2, 0x5, 0x0, 0x0, 0x0)
	/Users/distiller/project/transform/optimizer.go:59 +0x29d
github.com/tinygo-org/tinygo/builder.Build(0x7ffeefbffa15, 0x6, 0x7ffeefbff9f8, 0xf, 0xc0002d0060, 0xc0065e5cc0, 0x0, 0x0)
	/Users/distiller/project/builder/build.go:85 +0x2a21
main.Build(0x7ffeefbffa15, 0x6, 0x7ffeefbff9f8, 0xf, 0xc0002240e0, 0x0, 0x12)
	/Users/distiller/project/main.go:97 +0xc5
main.main()
	/Users/distiller/project/main.go:806 +0x192f

rax    0x104
rbx    0x2
rcx    0x7ffeefbff3f8
rdx    0x7100
rdi    0x8dc5f08
rsi    0x710100007200
rbp    0x7ffeefbff490
rsp    0x7ffeefbff3f8
r8     0x0
r9     0xa0
r10    0x0
r11    0x202
r12    0x8dc5f08
r13    0x16
r14    0x710100007200
r15    0x164f6dc0
rip    0x7fff64290ce6
rflags 0x203
cs     0x7
fs     0x0
gs     0x0
```

aykevl, 2020-05-08:
Well that's strange, it appears to hang in LLVM itself. I'll need to dig deeper into this.

aykevl, 2020-05-09:
When I try to compile this (using `tinygo build -o "./dist/lib.wasm" -target wasm ./wasm`), I get the following error:

```
# github.com/neophob/ganother-world/anotherworld
anotherworld/polygon.go:37:18: 0xFFFF0000 (untyped int constant 4294901760) overflows int
anotherworld/polygon.go:38:18: 0xFFFF0000 (untyped int constant 4294901760) overflows int
```

neophob, 2020-05-10:
jep, i twmp workaround this using 7FFFF0000

deadprogram, 2021-02-21:
Is this still an issue with the latest TinyGo dev branch?

deadprogram, 2024-05-05:
Closing due to age, please re-open if needed. Thanks!

closed 2024-05-05 by deadprogram