tinygo-org/tinygo #1087: Trying to implement `reflect.New()`

lei-april, 2020-05-08:
Hi,

I'm trying to implement the missing function `reflect.New()` in TinyGo, and here's my minimal patch:
```diff
diff --git a/src/reflect/value.go b/src/reflect/value.go
index a1c6937..6e953da 100644
--- a/src/reflect/value.go
+++ b/src/reflect/value.go
@@ -650,8 +650,16 @@ func Zero(typ Type) Value {
 	panic("unimplemented: reflect.Zero()")
 }
 
+//go:linkname alloc runtime.alloc
+func alloc(size uintptr) unsafe.Pointer
+
+func PtrTo(t Type) Type {
+	return (t << 5) + Type((Ptr-19)<<1) + 1
+}
+
 func New(typ Type) Value {
-	panic("unimplemented: reflect.New()")
+	data := alloc(typ.Size())
+	return Value{PtrTo(typ), data, 0}
 }
 
 type funcHeader struct {
```

When testing it with the following sample code:
```go
package main

import "reflect"

type FooInterface interface {
	Hello()
}

type Foo struct {
}

func (f *Foo) Hello() {
	println("Hello")
}

func main() {
	foo := reflect.New(reflect.TypeOf(Foo{})).Interface().(FooInterface)
	foo.Hello()
}
```

I got the error: `panic: runtime error: type assert failed`

However, if I change the sample code a little bit, it then works fine:
```go
func touch(i interface{}) {
}

func main() {
	touch(&Foo{}) // ???
	foo := reflect.New(reflect.TypeOf(Foo{})).Interface().(FooInterface)
	foo.Hello()
}
```

It seems that somehow I need to make the compiler see a conversion from `*Foo` to `interface{}`.

Do you have any insights on this? Thanks.

jaddr2line, 2020-05-08:
Quick explanation of what is happening:

Type asserts are implemented by generating a switch statement. TinyGo tries to compute a list of all implementing types, and then generates a type switch across them.

The way that TinyGo gets this list of all implementing types is by looking at every interface creation operation. In this case, you are synthesizing a type at runtime, and therefore the compiler does not discover the implementation.
So it looks like the compiler will also need to add cases for:
1. a pointer to a type which has been converted to an interface
2. all fields of types which have been converted to an interface

Other synthesized types cannot implement non-empty interfaces, so we do not need any gigantic changes to how we handle this yet.

deadprogram, 2021-12-06:
Will we be able to get back to this PR soon?

dgryski, 2021-12-06:
`reflect.New` has been implemented and the above tests case works.

```
~/go/src/github.com/dgryski/bug $ cat bug.go
package main

import "reflect"

type FooInterface interface {
	Hello()
}

type Foo struct {
}

func (f *Foo) Hello() {
	println("Hello")
}

func main() {
	foo := reflect.New(reflect.TypeOf(Foo{})).Interface().(FooInterface)
	foo.Hello()
}
~/go/src/github.com/dgryski/bug $ ~/go/src/github.com/tinygo-org/tinygo/build/tinygo run bug.go
Hello
```

closed 2021-12-07 by deadprogram