tinygo-org/tinygo #1103: build on ubuntu 20 fails

gshields, 2020-05-12:
with :
 clang --version
clang version 10.0.0-4ubuntu1 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
 uname -a
Linux 6540 5.4.0-29-generic #33-Ubuntu SMP Wed Apr 29 14:32:27 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
go version 
go version go1.14.2 linux/amd64

verified that clang is working to compile other programs 

Downloaded tinygo source

~/go/tinygo$ make
Fetch and build LLVM first by running:
  make llvm-source
  make llvm-build

Make fails to compile tinygo version as of 5/11/20

aykevl, 2020-05-12:
What is the error you are getting? (I'm assuming in `make llvm-build`).

aykevl, 2020-05-12:
If you mean that this is the error:

```
~/go/tinygo$ make
Fetch and build LLVM first by running:
make llvm-source
make llvm-build
```

Then you need to do what the error message says: run `make llvm-build`.

However, I think what you really want is not `make` but `go install`.

gshields, 2020-05-12:
tried this:
 sudo dpkg -i tinygo_0.13.1_amd64.deb
[sudo] password for ghs:
Selecting previously unselected package tinygo.
(Reading database ... 222884 files and directories currently installed.)
Preparing to unpack tinygo_0.13.1_amd64.deb ...
Unpacking tinygo (0.13.1) ...
Setting up tinygo (0.13.1) ...
ghs@6540:~$ tinygo
tinygo: command not found
I looked in /usr/local and there was no tinygo
tried to uninstall package and found it was not install
dpkg -P tinygo_0.13.1_amd64
dpkg: warning: ignoring request to remove tinygo_0.13.1_amd64 which isn't
installed

Then I tried to install using git source since this was on the instruction
git clone --recursive https://github.com/tinygo-org/tinygo.git
cd tinygo

I follow your suggestion to use go and I used go get and it worked!
Thanks

aykevl, 2020-05-12:
> ```
> ghs@6540:~$ tinygo
> tinygo: command not found
> ```
> I looked in /usr/local and there was no tinygo

I tried installing myself and a `/usr/local` did appear.
You should also add `/usr/local/tinygo/bin` to your `$PATH` variable:

    export PATH=$PATH:/usr/local/tinygo/bin

> tried to uninstall package and found it was not install
> ```
> dpkg -P tinygo_0.13.1_amd64
> dpkg: warning: ignoring request to remove tinygo_0.13.1_amd64 which isn't installed
> ```

This is not how you remove a package using `dpkg`. Remove it this way:

```
dpkg -P tinygo
```

You can also use `apt-get remove tinygo`.

> I follow your suggestion to use go and I used go get and it worked!
> Thanks

I'm glad you got it working!

closed 2020-05-12 by aykevl

gshields, 2020-05-13:
So I tried the correct command
dpkg -P tinygo
(Reading database ... 225765 files and directories currently installed.)
Removing tinygo (0.13.1) ...
dpkg: warning: while removing tinygo, directory '/usr/local' not empty so
not removed

dpkg apparently did not install using ubuntu 20.04

aykevl, 2020-05-14:
>     dpkg: warning: while removing tinygo, directory '/usr/local' not empty so not removed
> 
> dpkg apparently did not install using ubuntu 20.04

It probably did install correctly. There error indicates that it could not remove `/usr/local`, which is normal. I had the same error with a working installation. `/usr/local` usually contains locally installed programs but dpkg will try to remove it anyway because no other package had any files there that were tracked by dpkg.

In any case, the original issue you were hitting has likely been fixed with #1107 which will be part of the next release.