tinygo-org/tinygo #1114: VS Code Support

seanaye, 2020-05-16:
similar to #465 

I'm just getting started with go and am having trouble integrating with VS Code. 

I'm on mac and have installed tinygo via homebrew.

According to docs I can run 

`export GOPATH=$GOPATH:/usr/local/Cellar/tinygo/0.13.1/libexec/tinygo`

This however does nothing for vscode import errors.

```
cannot find package "machine" in any of:
	/usr/local/Cellar/go/1.14.2_1/libexec/src/machine (from $GOROOT)
	~/go/src/machine
```

I'm assuming I need to symlink `/usr/local/Cellar/tinygo/0.13.1/libexec/tinygo` into `~/go/src/machine` 

 Does anyone know how to fix these errors in vscode? What is the recommended approach? As I said i'm inexperienced with golang,

aykevl, 2020-05-16:
It appears that the GOPATH wasn't picked up VS Code. Remeber that `export` only works in the current shell, it is not system-wide. You might need to change some settings in VS Code itself to get this to work.

This may be helpful: https://github.com/Microsoft/vscode-go/wiki/GOPATH-in-the-VS-Code-Go-extension

seanaye, 2020-05-16:
@aykevl Thanks for quick reply

I have added tinygo to my .bash_profile

`export GOPATH=$GOPATH:/usr/local/Cellar/tinygo/0.13.1/libexec/tinygo`

`go env GOPATH`
returns 
`/usr/local/Cellar/tinygo/0.13.1/libexec/tinygo`

after closing and restarting vscode `Go: Current GOPATH` now returns the above path.

This works but is it the recommended way of writing code in go?

Is there a way to get type hints for my specific board? its the Arduino nano 33 IoT

looking at sample code

```go
package main

import (
	"machine"
	"time"
)

func main() {
	led := machine.LED
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
}
```
I get type hints for machine.PinConfig but not for machine.LED

aykevl, 2020-05-16:
> This works but is it the recommended way of writing code in go?
> 
> Is there a way to get type hints for my specific board? its the Arduino nano 33 IoT

You will also need to configure build tags. I think you can use `go.buildTags` for that. To get the list of build tags, run:

    tinygo info arduino-nano33

I know this is a bit involved. I'm hoping that we can properly fix this in the future (#941 is a part of that). Until then, this should work.

seanaye, 2020-05-17:
Great thanks for all your help. I'm excited to write some Go!

closed 2020-05-17 by seanaye