tinygo-org/tinygo #1115: Error WASM 'sync/atomic.LoadPointer' is not a Function

brainlulz, 2020-05-16:
I'm just getting started with Go and WASM.

While uploading an image and decoding it on the WASM side, I have this error on my browser console:
`LinkError: import object field 'sync/atomic.LoadPointer' is not a Function`.

I don't have it when using the command `GOARCH=wasm GOOS=js go build -o lib.wasm main.go`.

I used the correct `wasm_exec.js` file for TinyGo and my compile command is `tinygo build -o lib.wasm -target wasm main.go`.

I checked and all packages that I'm using seems supported.

Here is the test repository: https://github.com/brainlulz/wasm-go-image
If you use the commange `GOARCH=wasm GOOS=js go build -o lib.wasm main.go` and change the wasm_exec.js` file it's working. 

Any idea where it's comming from ?

jaddr2line, 2020-05-16:
I think your issue should be resolved with https://github.com/tinygo-org/tinygo/pull/1113

aykevl, 2020-05-17:
Yes, most likely. With that PR, atomics will be implemented for all supported platforms including WebAssembly.

brainlulz, 2020-05-17:
Awesome, thanks! I will retry with TinyGo when the fix is released :)

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram