tinygo-org/tinygo #1118: Using a custom CDC config

andrewrynhard, 2020-05-17:
Hello,

I am fairly new to the world of microcontrollers and I am trying to use the Arduino Nano 33 IoT as a keyboard. The documentation of this board says that this is possible. From what I could gather, there are two different ways to get a computer to recognize a USB device as a keyboard. Using HID or CDC. The Nano 33 IoT seems to only support the latter. I Have been playing with TinyGo, but there doesn't seems to be a way to use a custom CDC config/(descriptor?) so that I can tell the computer that the board should be classified as a keyboard. I would appreciate any kind of help, and I am happy to contribute in any way I can if I need to.

deadprogram, 2020-05-18:
USB CDC stands for "Communication Device Class". Please see https://en.wikipedia.org/wiki/USB_communications_device_class

USB HID stands for "Human Interface Device" Please see https://en.wikipedia.org/wiki/USB_human_interface_device_class

These are both USB protocols, but only HID has anything to do with keyboards. CDC is for USB modems, network adaptors, etc.

In order to use TinyGo to create a USB HID keyboard you would need to implement support for HID protocol for SAMD21 based MCUs like the Arduino Nano33.

andrewrynhard, 2020-05-18:
@deadprogram thanks for the clarification. Mind pointing me in the right direction in the tinygo code where I might start looking into that?

deadprogram, 2020-05-18:
The current USB CDC code that applies to all implementation is here: https://github.com/tinygo-org/tinygo/blob/master/src/machine/usb.go

The code specific for USB to the SAMD21 family starts here: https://github.com/tinygo-org/tinygo/blob/master/src/machine/machine_atsamd21.go#L1257

bgould, 2020-05-18:
BTW if we do implement HID I have already started porting the TMK framework to Go, found here: https://github.com/bgould/tinygo-model-m

Constants for keyboard HID usage are here (not all of these codes are 100% standard, some are TMK-specific): https://github.com/bgould/tinygo-model-m/blob/master/keyboard/keycodes/keycodes.go

If we get a prototype of HID working it should be usable to test on a keyboard by implementing this interface: https://github.com/bgould/tinygo-model-m/blob/master/keyboard/keyboard.go#L8

andrewrynhard, 2020-05-18:
@bgould nice. This is more or less what I was planning on doing. Wanted to have some fun and create firmware using tinygo that does more or less the same thing as QMK. I will take a look at what you have once I can start to wrap my head around things some more.

bgould, 2020-08-06:
I think I'd like to try to implement this for inclusion in the next release, at least HID keyboard and HID mouse on SAMD chips.  I can do my best to split out the portable parts as @deadprogram referenced in the CDC examples.

aykevl, 2020-10-27:
I think the first step would be to separate the USB code into a new package. It is currently in the machine package because it provides a serial output, but to make it more useful I think it would be best to move it to a new package that is used by the machine package.

sago35, 2023-02-21:
USB HID is supported via `TinyGo 0.24`.
https://github.com/tinygo-org/tinygo/releases/tag/v0.24.0

closed 2023-02-21 by sago35