tinygo-org/tinygo #1119: Timing inaccuracy on BBC Microbit (nrf51)

harry1453, 2020-05-17:
When running this code on the BBC micro:bit

```go
package main

import (
	"machine"
	"time"
)

func main()  {
	pin := machine.P0
	pin.Configure(machine.PinConfig{Mode: machine.PinOutput})
        // Drive pin low for a bit to get clearer start / stop points for Sleep call
	pin.Low()
	time.Sleep(100 * time.Millisecond)
	pin.High()
	time.Sleep(time.Second)
	pin.Low()
}
```

```
tinygo flash --target=microbit
```

and looking at pin P0, I get this:

[image]

Showing that the `time.Sleep` call only actually sleeps for 890ms... This is actually noticeable if you try the blink program, it blinks slightly too fast.

When I run this very similar Rust program:

```rust
#![no_main]
#![no_std]

use panic_abort as _;

use microbit::hal::delay::Delay;
use microbit::hal::prelude::*;
use nrf51_hal::gpio::GpioExt;
use nrf51_hal::gpio::gpio::Parts;

use cortex_m_rt::entry;

#[entry]
fn main() -> ! {
    if let Some(peripherals) = microbit::Peripherals::take() {
        let gpio: Parts = peripherals.GPIO.split();
        let mut delay = Delay::new(peripherals.TIMER0);
        let mut led = gpio.pin3.into_push_pull_output();
        // Drive pin low for a bit to get clearer start / stop points for Sleep call
        led.set_low();
        delay.delay_ms(100u16);
        led.set_high();
        delay.delay_ms(1000u16);
        led.set_low();
    }

    loop {
        continue;
    }
}
```

I get the correct delay length:

[image]

I would prefer to use TinyGo, but this is a significant timing inaccuracy... Please could someone take a look at this?

aykevl, 2020-05-21:
Thank you for reporting this. This is a long standing TODO in the code (that is in fact outdated as it refers to the incorrectly-named `tickMicros` above):

https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/src/runtime/runtime_nrf.go#L67

Properly fixing this needed a bit more work, which even surfaced issues in other chips. PR https://github.com/tinygo-org/tinygo/pull/1127 should fix this issue. It also gets rid of the incorrectly named `tickMicros` (which was in nanoseconds, not microseconds).

harry1453, 2020-06-01:
Thanks for your response! Sorry for my delay in replying. I did the following (using Docker to have a reproducible environment as I'm running Windows):

* `git clone https://github.com/tinygo-org/tinygo && cd tinygo`
* `git checkout origin/dev`

At this point, `git log --oneline` reported the following as HEAD: `3c31a31 (HEAD, origin/dev) builder: use newer version of gohex`

* `docker build .` which produced image `19aa8a2d4b4b`

* `docker run -v R:\tmp\:/src 19aa8a2d4b4b tinygo build -o /src/build.hex -target=microbit /src/main.go`

I ran this using the exact same program as before.

This produced build.hex which, upon flashing to the device, produced this:

[image]

So it seems to have been improved but not fully resolved...

BTW, Thanks for the time and effort you're putting into this and this awesome project!

deadprogram, 2020-08-07:
This was released with v0.14.0 so now closing. Please reopen if needed. Thanks!

closed 2020-08-07 by deadprogram