tinygo-org/tinygo #1120: Is there any way to import a function from another module in wasm explicitly?

moznion, 2020-05-19:
Hi,

I'm trying to import a function from another module in wasm by the following code:

```go
package main

func doSomething() // <- I'd like to import this

func main() {}
```

and then I built a wasm file by `tinygo build -o main.wasm -target wasm ./main.go`, but generated wasm file seems it doesn't contain an `import` instruction.

Following wat code tip is converted from the wasm file:

```
(module
  (type (;0;) (func (param i32)))
  (type (;1;) (func (param i32 i32 i32 i32) (result i32)))
  (type (;2;) (func))
  (type (;3;) (func (param i32 i32 i32) (result i32)))
  (type (;4;) (func (param i32 i32)))
  (import "wasi_unstable" "fd_write" (func $fd_write (type 1)))
  (func $__wasm_call_ctors (type 2))
...
```

(I translated this by [WebAssembly/wabt](https://github.com/WebAssembly/wabt))

That only contains `(import "wasi_unstable" "fd_write" (func $fd_write (type 1)))`.
What I want is `(import "env" "doSomething" (func $doSomething (type 2)))` instruction, is this possible?

aykevl, 2020-05-19:
Yes, you already did declare an import. However, if you're not calling the function the compiler will optimize it away (as there is no point in importing a function that will not be used). Try calling it (for example, from `main`) to get the import to work.

That said, in general you should mark such imports with `//export`. That's a bit of a misnomer (coming from CGo) but it does declare an import following the C abi instead of the TinyGo internal ABI.

```go
package main

//export doSomething
func doSomething()

func main() {
    doSomething()
}
```

moznion, 2020-05-20:
I had forgotten the compiler optimization!
That code works fine as I expected.

I appreciate your answer!

closed 2020-05-20 by moznion