tinygo-org/tinygo #1123: How can I avoid importing `wasi_unstable/fd_write` in wasm?

moznion, 2020-05-20:
Hello,

When I compile a wasm with an empty go code that likes the following,

```go
package main

func main() {}
```

 the generated wasm always imports `wasi_unstable/fd_write` like the following wat:

```wat
(module
  (type (;0;) (func (param i32)))
  (type (;1;) (func (param i32 i32 i32 i32) (result i32)))
  (type (;2;) (func))
  (type (;3;) (func (param i32 i32 i32) (result i32)))
  (type (;4;) (func (param i32 i32)))
  (import "wasi_unstable" "fd_write" (func $fd_write (type 1)))
 ...
```

It seems [runtime_wasm.go](https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/src/runtime/runtime_wasm.go) contains the following code to import that function:

https://github.com/tinygo-org/tinygo/blob/a9ba6ebad9ad85cacd9674a1af9229489a811f68/src/runtime/runtime_wasm.go#L17-L19

so does this mean I cannot avoid importing `wasi_unstable/fd_write`? Or is there another way to do that?

aykevl, 2020-05-20:
This is necessary for the runtime, to print `println` and panic messages for example. It looks like these may also be generated during runtime initialization. What are you trying to do that you need such tight control over the output?

moznion, 2020-05-20:
Thank you for your response!

I've tried to run the wasm that generated by tinygo on rust wasm runtime, but the runtime doesn't provide `wasi_unstable/fd_write` as default, so it raises an error like the following:

```
{"errorType":"failure::ErrorMessage","errorMessage":"failure::ErrorMessage: failure::ErrorMessage: Failed to link module\n\nCaused by:\n    Link error: wasi_unstable/fd_write: unknown import function: function not provided"}
```

So I'm curious about how to avoid this problem (Of course, it would be better to provide that function on rust runtime).

aykevl, 2020-05-20:
Does the Rust wasm runtime you mention support WASI? Because this is a standard WASI call. Non-browser usage of TinyGo requires a runtime that supports WASI.

moznion, 2020-05-21:
I execute the wasm in a sandbox so it limits the function to integrate with WASI; i.e. it intents to block to access to the OS layer. That is a reason what I would like to bypass importing it.
Anyway, as you mentioned, it must support WASI to execute wasm that generated by tinygo, is it true?

moznion, 2020-05-21:
Coincidentally, I found a dirty-way to don't import `wasi_unstable/fd_write`:

```go
package main

//export fd_write
func fd_write() (errno uint) {
	return 0
}

func main() {
}
```

If it overwrites the function by `//export`, the generated wasm doesn't contain `import` instruction.

aykevl, 2020-05-21:
> I execute the wasm in a sandbox so it limits the function to integrate with WASI; i.e. it intents to block to access to the OS layer. That is a reason what I would like to bypass importing it.

The `fd_write` call can indeed be used for OS access, but in this case it is only used for writing to stdout. There is little harm in doing that.

>  Anyway, as you mentioned, it must support WASI to execute wasm that generated by tinygo, is it true?

It must support that one call to be able to print to stdout, yes.

> Coincidentally, I found a dirty-way to don't import `wasi_unstable/fd_write`

Well yes that is a way to work around it...

moznion, 2020-05-21:
I understand. I'll consider making the sandbox supports WASI, somehow.
Thank you for your support!

closed 2020-05-21 by moznion