tinygo-org/tinygo #1130: Is ICE40 FPGA support feasible?

wade-welles, 2020-05-23:
Ive been working with tinygo on AVR's and really like the process compared to the alternatives. I have been doing research on if it is possible to generate Bitstreams from LLVM and it appears to be possible, but since I'm new to this I don't know if this line of research will pan out. 

How feasible is extending tinyGo to support ICE40 branch of FPGAs (the ones that already have open source toolchain)? Because the current toolchain is way too clunky and if I can avoid learning verilog I would like that. 

Thanks

aykevl, 2020-05-23:
I honestly don't know.
The easy route would be to put a small processor core on the ICE40 (such as a small RISC-V core) and run a program on that. But it sounds like that's not what you're after.
As far as I'm aware, compiling regular programs directly to FPGAs is very difficult, and there aren't any good solutions yet. But maybe things have improved since the last time I looked at it.

I think the first step would be to get _anything_ to run on an ICE40 that is a regular program (in C or Rust). If that works, we can see how feasible it is in TinyGo. But otherwise I don't think this is the kind of device that TinyGo targets.

jaddr2line, 2020-05-23:
This is possible, but there would be a lot of restrictions. You would not be able to use dynamically allocated memory or the scheduler (therefore no goroutines or channels); you would also need to come up with a meaning for panic, and a lot of things would be extremely inefficient.

At one point I did investigate high level synthesis using a subset of Go (including channels), and concluded that you could probably make something decently efficient by transforming goroutines into state machines, and channels into flip-flop-like structures (unfortunately I do not really have the time to spare to pursue this).

TinyGo is probably not an ideal choice for this use case. I had used this to learn some verilog basics: http://www.asic-world.com/verilog/veritut.html

but verilog really is somewhat weird to work with.

wade-welles, 2020-05-24:
Thanks, I appreciate the insight, I got excited because I was seeing that people in the past have tried to do C++ LLVM solution to build bitstreams but C++ being sequentially focused was not a good fit for FPGA development. 

So if I understand correctly you are saying the most likely best path is a subset of the compiler with its own collection of special stdlibraries for the purpose of providing all those unique FPGA features and since those features are quite different from the typical AVR/ARM/X86 style processor that it would not make sense for TinyGo to fit this objective. 

Okay, well I still plan seeing if any of my work in TinyGo will be useful to the core project. I have been building libraries focusing on the chips pinout, generating them automatically from dataseets to generate Go libraries that contain a lot of the useful information for working on a specific chip. Then you can define the breakout board for even better control. I understand that the machine stdlibrary is intended to be light as weight as possible, these libraries are meant to provide an additional layer ontop of that to have access to all the details of the chip and the board without having to rewrite it each time. 

Thanks again.

closed 2020-05-24 by wade-welles

wade-welles, 2020-05-24:
Thanks, I appreciate the insight, I got excited because I was seeing that people in the past have tried to do C++ LLVM solution to build bitstreams but C++ being sequentially focused was not a good fit for FPGA development. 

So if I understand correctly you are saying the most likely best path is a subset of the compiler with its own collection of special stdlibraries for the purpose of providing all those unique FPGA features and since those features are quite different from the typical AVR/ARM/X86 style processor that it would not make sense for TinyGo to fit this objective. 

Okay, well I still plan seeing if any of my work in TinyGo will be useful to the core project. I have been building libraries focusing on the chips pinout, generating them automatically from dataseets to generate Go libraries that contain a lot of the useful information for working on a specific chip. Then you can define the breakout board for even better control. I understand that the machine stdlibrary is intended to be light as weight as possible, these libraries are meant to provide an additional layer ontop of that to have access to all the details of the chip and the board without having to rewrite it each time. 

Thanks again. 

> I honestly don't know.
> The easy route would be to put a small processor core on the ICE40 (such as a small RISC-V core) and run a program on that. But it sounds like that's not what you're after.
> As [...]

Thanks for mentioning this; This post is for me but also for all the people who will have this same idea and question and this will be here to help people see what options are available. 

I was not originally going this route, but when doing my research I was discovering that this is also an approach to take; basically emulating one ore more ARM64 processors or RISC V etc.. So concurrency is achieved by emulating more than one on an FPGA device and then loading the tinyGo firmware onto those emulated deviced. 

Since my plan was to build a firmata style interface for the FPGA anyways, something like this may be what I eventually do. 

Do you have any experience doing this or you just know about it from doing research?

aykevl, 2020-05-24:
> Do you have any experience doing this or you just know about it from doing research?

I have never worked with FPGAs and don't know much about them. All I know is that the programming model is very different.

Because most microcontrollers have a single core and have little hardware support for task switching (if any), everything is done in software. I know there are people who have tried to lower Go style primitives (goroutines, channels) directly to what FPGAs support and maybe that's a better approach. You are free to try it, of course.

trevor403, 2020-05-25:
Related issue #948