tinygo-org/tinygo #1132: Have you considered using comments prefixing the source files like Cgo, and architecture definitins to declare compile flags needed instead of separate JSON files?

wade-welles, 2020-05-24:
Have you considered using comments prefixing the source files like Cgo, and architecture definitins to declare compile flags needed instead of separate JSON files?

Are the JSON files a format specifically used by other AVR firmware compilers or is it unique to TinyGo? Because currently they inherit, and so some of them require opening 4 files to see the actual full set of things that will be required to correctly compile the firmware. 

I was writing tools to build my Makefiles automatically from the JSON files, so I would have a place where its all compiled together and I can review and modify (because I''m experimenting with adding more supported chips). 

And after staring at it for a while, I realized these may be the most consistent with other aspects of Go if these compiler options and perhaps even the flash options were defined in the header or prefixing the source code. 

I started experimenting using AST library provided by the Go compiler to do just that, and don't want to assume you would agree and thought I would pitch this idea and see what everyone else thought about the concept.

aykevl, 2020-05-24:
The JSON format is a custom format, very loosely based on the [Rust target specification](https://github.com/rust-lang/rfcs/blob/master/text/0131-target-specification.md). You can get some options using a command like this:

    tinygo info microbit

While the information could theoretically be stored in Go files directly, you would still have to specify the target somehow, including a list of build tags. So how would you specify the exact board when the information is stored in Go source files?

> I was writing tools to build my Makefiles automatically from the JSON files, so I would have a place where its all compiled together and I can review and modify (because I''m experimenting with adding more supported chips).

Why do you need to build Makefiles?

deadprogram, 2021-04-22:
Closing due to inactivity. Please reopen if needed. Thank you!

closed 2021-04-22 by deadprogram